balena-io/etcher #2713: Writer process ended unexpetedly

vamp111, 2019-03-27:
I get this error when trying to write on a brand new SD, after this the card becomes unrecognizable, windows doesn't see the card, can't format it! I've tried with 2nd card, which was used many times, got the same error, now again the card is unrecognizable. Anyone had this problem ? a way to restore my SDcard ?

Etcher ver. 1.5.5 on Windows 10

lurch, 2019-03-27:
> Anyone had this problem ?

Yeah, looks like it https://github.com/balena-io/etcher/search?q=Writer+process+ended+unexpectedly&type=Issues

thundron, 2019-04-03:
@vamp111 Does it fail right away when trying to flash or does it stop at a certain step? Try also with the latest `1.5.19`

vamp111, 2019-04-04:
it doesnt start writing, crashes about 3-5 sec. after I press the "flash"button. I tryed an older version as i read reccomendations and it worked just fine, also i managed to save my SD card, recovering it using an old method. Thx

zaxhalmon, 2019-04-07:
Having this very same issue happen to two different Samsung 32GB SD Cards, rendering them useless. Using Version 1.5.19.

vamp111, 2019-04-08:
> Having this very same issue happen to two different Samsung 32GB SD Cards, rendering them useless. Using Version 1.5.19.

follow this guide here, so you can recover your cards at least: https://www.youtube.com/watch?v=hYwjMmJ4qk0

vamp111, 2019-04-10:
@thundron no nothing, and yet again I still dont know what happened. Anyway the recovery methods didnt worked, so I figured one by myself you can see the vid. one comment above yours ;)

thundron, 2019-04-10:
> along with others you might already know

We're still investigating the issue, looks like it happens mostly on Windows for now

vamp111, 2019-04-10:
@thundron this was on Windows 10, I never had issues flashing cards on Ubuntu 16.04 before

closed 2019-04-10 by vamp111