balena-io/etcher #2715: Etcher 1.5.18 not running properly under Tumbleweed

fleamour, 2019-03-28:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

```

Uncaught Error: /tmp/.org.chromium.Chromium.AvGBqQ: invalid ELF header
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:180:18)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
    at Module.require (internal/modules/cjs/loader.js:658)
    at require (internal/modules/cjs/helpers.js:20)
    at Object.<anonymous> (/tmp/.mount_balenavNY3eP/resources/app.asar/node_modules/xxhash/lib/xxhash.js:4)
    at Object.<anonymous> (/tmp/.mount_balenavNY3eP/resources/app.asar/node_modules/xxhash/lib/xxhash.js:52)
    at Module._compile (internal/modules/cjs/loader.js:711)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
    at Module.require (internal/modules/cjs/loader.js:658)
```

lurch, 2019-03-28:
See also https://github.com/balena-io/etcher/issues/2712#issuecomment-477288405

zvin, 2019-03-29:
Fixed in 1.5.19

closed 2019-03-29 by zvin