balena-io/etcher #2722: Writer process ended unexpectedly

pb1uk, 2019-04-01:
I'm receiving at error "the writer process ended unexpectedly. Please try again, and contact the Etcher team if the problem persists". I'm trying to copy the image (link below) to a Samsung 32GB 95MB/s Memory Evo Plus Micro SD Card (microSDHC UHS-I Card)

Can anyone advise on what the issue may be please? Have also tried v1.8.1 and 1.2.1 with the same result.

- **Etcher version:** 1.9.1
- **Operating system and architecture:** Windows 10 Pro x64
- **Image flashed:** http://interaction.gold.ac.uk/naturewatch-cam-v0p4p2.img
- **Do you see any meaningful error information in the DevTools?**

thundron, 2019-04-03:
@pb1uk That version definitely looks *too* old though, I advise to try with the most recent `1.5.19`

zvin, 2019-04-03:
@pb1uk there's no `1.9.1` or `1.8.1` version.
Does this happen with all disk images or only this one?