balena-io/etcher #2723: Creates 2MB partition in middle of 7GB USB stick and tries to install into it

skirsch, 2019-04-02:
- **Etcher version:** 
1.5.19  64 bit
- **Operating system and architecture:** 
Windows 10
- **Image flashed:** 
Ubuntu
- **Do you see any meaningful error information in the DevTools?** 

This bug has been happening for over  a year it seems with people saying it destroyed their USB stick. Disk Management shows Etcher creates a 2MB partition right in the middle of the disk, splitting the 7G partition in two pieces so now there are 3 partitions. THen it attempts to install on the 2MB partition. I've formatted the disk as FAT32 before running Etcher. Really annoying and 100% repeatable.

lurch, 2019-04-02:
AFAIK this is just a quirk of the way that Ubuntu disk images set up their partitions and/or the way that Windows 10 interprets the unusual partitioning scheme. Nothing to do with Etcher itself.

More info at https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos and https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

zvin, 2019-04-03:
Etcher does not create partitions, it only copies the raw content of a disk image to a medium.
The raw content may be anything, including partition tables or partition types that are not recognized by your system.

closed 2019-04-03 by zvin