balena-io/etcher #2730: Balena Etcher do not detect new version for Windows

OTLabs, 2019-04-13:
- **Etcher version:** 1.5.5 
- **Operating system and architecture:** Windows 7 64 bit
- **Image flashed:** no
- **Do you see any meaningful error information in the DevTools?** no

Download the React DevTools for a better development experience: https://fb.me/react-devtools
You might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
C:\Users\otlabs\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Warning: The "extend" API will be removed in the upcoming v4.0 release. Use styled(StyledComponent) instead. You can find more information here: https://github.com/styled-components/styled-components/issues/1546
console.(anonymous function) @ C:\Users\otlabs\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58
C:\Users\otlabs\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.5, Type = nsis
C:\Users\otlabs\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Sat Apr 13 2019 12:06:50 GMT-0500 (Hora de verano central (México)) Application start ({"packageType":"nsis","version":"1.5.5","applicationSessionUuid":"50140ed5-8d6a-4606-8208-c8bac358d3a7"})
C:\Users\otlabs\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Sat Apr 13 2019 12:06:50 GMT-0500 (Hora de verano central (México)) Checking for updates ({"currentVersion":"1.5.5","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false,"applicationSessionUuid":"50140ed5-8d6a-4606-8208-c8bac358d3a7"})
C:\Users\otlabs\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Sat Apr 13 2019 12:06:52 GMT-0500 (Hora de verano central (México)) Update notification skipped ({"reason":"Latest version","applicationSessionUuid":"50140ed5-8d6a-4606-8208-c8bac358d3a7"})
C:\Users\otlabs\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Sat Apr 13 2019 12:06:52 GMT-0500 (Hora de verano central (México)) SafeWebview loaded ({"event":{"fromCache":false,"headers":{"Accept":"text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8","Upgrade-Insecure-Requests":"1","User-Agent":"Mozilla/5.0 (Windows NT 6.1; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) balena-etcher/1.5.5 Chrome/66.0.3359.181 Electron/3.1.3 Safari/537.36"},"id":10,"method":"GET","referrer":"","resourceType":"mainFrame","responseHeaders":{"Accept-Ranges":["bytes"],"Cache-Control":["public, max-age=0"],"Connection":["keep-alive"],"Content-Encoding":["gzip"],"Content-Type":["text/html; charset=UTF-8"],"Date":["Sat, 13 Apr 2019 17:06:48 GMT"],"Etag":["W/\"3a56d-16a116f6450\""],"Last-Modified":["Fri, 12 Apr 2019 12:04:34 GMT"],"Server":["Cowboy"],"Transfer-Encoding":["chunked"],"Vary":["Accept-Encoding"],"Via":["1.1 vegur, 1.1 vegur"]},"statusCode":200,"statusLine":"HTTP/1.1 200 OK","timestamp":1555175212249.247,"url":"https://www.balena.io/etcher/success-banner/?etcher-version=1.5.5&api-version=2"},"applicationSessionUuid":"50140ed5-8d6a-4606-8208-c8bac358d3a7","flashingWorkflowUuid":"ac79577c-6cec-43da-87ea-55704423f1ff"})

I have Balena Etche v1.5.5, it is unable to detect any new versions. Same time if I use ETCHER_FAKE_S3_LATEST_VERSION=1.5.6 the update alert is shown.

thundron, 2019-04-18:
@OTLabs Sorry but I'm not sure what the issue is :/ you can directly download the package from the website here https://balena.io/etcher 
Is that what you're after?

OTLabs, 2019-04-18:
@thundron Thank you for taking a look at the issue I have! I was expecting that if I run Balena Etcher v1.5.5 it will detect new version available and will show a notification. The notification is not shown. Right now I just started Balena Etcher v1.5.5 and there was no any notification windows despite the main web site already has v1.5.24. Hope I was able to clarify the behavior I would like to see.