balena-io/etcher #2731: Etcher can't detect both of my USBs (Ubuntu-AppImage)

yagizhan49, 2019-04-13:
- **Etcher version:** 1.5.24
- **Operating system and architecture:** Ubuntu 18.04 i386
- **Image flashed:** ubuntu-mate-18.10-desktop-amd64.iso
- **Do you see any meaningful error information in the DevTools?** I can't even launch DevTools.

I tried to burn Ubuntu MATE to my USB using Etcher, but I got an error message with the title of "Connect a drive!".

Unetbootin was able to detect both of my USBs while Etcher couldn't detect both.
And yes, I tried with different USB ports.

I burned them using Unetbootin and I wasn't even able to launch the Live USB. My laptop just skipped the USB and continued with a typical startup. So I want to try Etcher.

lurch, 2019-04-14:
Possibly another [drivelist](https://github.com/balena-io-modules/drivelist/) bug?
What output do you get when you run
```bash
lsblk --bytes --all --json --paths --output-all
```
in a terminal? (obviously with both of your USB sticks plugged in)

yagizhan49, 2019-04-14:
Here is the cropped output (removed the unnecessary part)

```
{
   "blockdevices": [
      {"name": "/dev/sda", "kname": "/dev/sda", "maj:mIn": "8:0", "fstype": null, "mountpoInt": null, "label": null, "uuId": null, "parttype": null, "partlabel": null, "partuuId": null, "partflags": null, "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": "WDC WD2500BEVT-6", "serIal": "WD-WX50A9943726", "sIze": "250059350016", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-sIze": "128", "type": "disk", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": "0x50014ee258baee50", "rand": "1", "pkname": null, "hctl": "0:0:0:0", "tran": "sata", "subsystems": "block:scsi:pci", "rev": "1A13", "vendor": "ATA     ", "zoned": "none",
         "children": [
            {"name": "/dev/sda1", "kname": "/dev/sda1", "maj:mIn": "8:1", "fstype": "ext4", "mountpoInt": "/", "label": null, "uuId": "a096aaaa-a66e-4866-86f4-02d574feb7f6", "parttype": "0x83", "partlabel": null, "partuuId": "4f5df836-01", "partflags": "0x80", "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": null, "serIal": null, "sIze": "250058113024", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-sIze": "128", "type": "part", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": "0x50014ee258baee50", "rand": "1", "pkname": "/dev/sda", "hctl": null, "tran": null, "subsystems": "block:scsi:pci", "rev": null, "vendor": null, "zoned": "none"}
         ]
      },
      {"name": "/dev/sdb", "kname": "/dev/sdb", "maj:mIn": "8:16", "fstype": "iso9660", "mountpoInt": null, "label": "Ubuntu-MATE 18.10 amd64", "uuId": "2018-10-17-22-46-16-00", "parttype": null, "partlabel": null, "partuuId": null, "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": "Cruzer Glide 3.0", "serIal": "4C530001320119117313", "sIze": "15669919744", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-sIze": "128", "type": "disk", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": null, "hctl": "4:0:0:0", "tran": "usb", "subsystems": "block:scsi:usb:pci", "rev": "1.00", "vendor": "SanDisk ", "zoned": "none",
         "children": [
            {"name": "/dev/sdb1", "kname": "/dev/sdb1", "maj:mIn": "8:17", "fstype": "iso9660", "mountpoInt": null, "label": "Ubuntu-MATE 18.10 amd64", "uuId": "2018-10-17-22-46-16-00", "parttype": "0x0", "partlabel": null, "partuuId": "18b86318-01", "partflags": "0x80", "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": null, "serIal": null, "sIze": "2103607296", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-sIze": "128", "type": "part", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": "/dev/sdb", "hctl": null, "tran": null, "subsystems": "block:scsi:usb:pci", "rev": null, "vendor": null, "zoned": "none"},
            {"name": "/dev/sdb2", "kname": "/dev/sdb2", "maj:mIn": "8:18", "fstype": "vfat", "mountpoInt": null, "label": "Ubuntu-MATE 18.10 amd64", "uuId": "B4D9-686A", "parttype": "0xef", "partlabel": null, "partuuId": "18b86318-02", "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": null, "serIal": null, "sIze": "2523136", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-sIze": "128", "type": "part", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": "/dev/sdb", "hctl": null, "tran": null, "subsystems": "block:scsi:usb:pci", "rev": null, "vendor": null, "zoned": "none"}
         ]
      },
      {"name": "/dev/sdc", "kname": "/dev/sdc", "maj:mIn": "8:32", "fstype": "iso9660", "mountpoInt": null, "label": "Ubuntu-MATE 18.10 amd64", "uuId": "2018-10-17-22-46-16-00", "parttype": null, "partlabel": null, "partuuId": null, "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": "TransMemory     ", "serIal": "C412F52D6C8DC1118002C7B9", "sIze": "15518924800", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-sIze": "128", "type": "disk", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": null, "hctl": "5:0:0:0", "tran": "usb", "subsystems": "block:scsi:usb:pci", "rev": "1.00", "vendor": "TOSHIBA ", "zoned": "none",
         "children": [
            {"name": "/dev/sdc1", "kname": "/dev/sdc1", "maj:mIn": "8:33", "fstype": "vfat", "mountpoInt": null, "label": "Ubuntu-MATE 18.10 amd64", "uuId": "52ED-5AD1", "parttype": "ebd0a0a2-b9e5-4433-87c0-68b6b72699c7", "partlabel": null, "partuuI [... cut ...] 584768", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-sIze": "128", "type": "part", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": "/dev/sdc", "hctl": null, "tran": null, "subsystems": "block:scsi:usb:pci", "rev": null, "vendor": null, "zoned": "none"}
         ]
      }
   ]
}
```

lurch, 2019-04-14:
Why have all the `i` characters in your output been converted to upper-case?
`"uuId"`, `"serIal"`, `"sIze"`, etc.

yagizhan49, 2019-04-14:
I'm not sure, that's what I got as response.

thundron, 2019-04-15:
That looks strange. It looks like only the keys (and not all of them) have uppercase `i`s while the values look correct

szb512, 2019-05-04:
Have you tried formatting the USB's?

queeup, 2019-08-21:
Same problem here. Non of my usb sticks detected. Tried format NTFS/exFAT/FAT32. No sign of recognize.

elementary OS, balenaEtcher-1.5.56-x64.AppImage

```console
 ~ $ blkid -v
blkid from util-linux 2.31.1  (libblkid 2.31.1, 19-Dec-2017)
```

```console
~ $ lsblk --bytes --all --json --paths --output-all
{
   "blockdevices": [
            {"name": "/dev/sdc", "kname": "/dev/sdc", "maj:mIn": "8:32", "fstype": null, "mountpoInt": null, "label": null, "uuId": null, "parttype": null, "partlabel": null, "partuuId": null, "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": "DataTraveler 2.0", "serIal": "000AEB91BC8CF9A1A5500DDF", "sIze": "4002910208", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-sIze": "128", "type": "disk", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": null, "hctl": "8:0:0:0", "tran": "usb", "subsystems": "block:scsi:usb:pci", "rev": "1.00", "vendor": "Kingston", "zoned": "none",
         "children": [
            {"name": "/dev/sdc1", "kname": "/dev/sdc1", "maj:mIn": "8:33", "fstype": "exfat", "mountpoInt": null, "label": "4GB", "uuId": "69AE-7D3C", "parttype": "0x7", "partlabel": null, "partuuId": "f662eea2-01", "partflags": "0x80", "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": null, "serIal": null, "sIze": "4001366016", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-sIze": "128", "type": "part", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": "/dev/sdc", "hctl": null, "tran": null, "subsystems": "block:scsi:usb:pci", "rev": null, "vendor": null, "zoned": "none"}
         ]
      },
      {"name": "/dev/sdd", "kname": "/dev/sdd", "maj:mIn": "8:48", "fstype": null, "mountpoInt": null, "label": null, "uuId": null, "parttype": null, "partlabel": null, "partuuId": null, "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": "DataTraveler G3 ", "serIal": "001372982FFBEBC145FC0290", "sIze": "7803174912", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-sIze": "128", "type": "disk", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": null, "hctl": "9:0:0:0", "tran": "usb", "subsystems": "block:scsi:usb:pci", "rev": "PMAP", "vendor": "Kingston", "zoned": "none",
         "children": [
            {"name": "/dev/sdd1", "kname": "/dev/sdd1", "maj:mIn": "8:49", "fstype": "exfat", "mountpoInt": null, "label": null, "uuId": "28B7-94BC", "parttype": "0x7", "partlabel": null, "partuuId": "b20e87e2-01", "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": null, "serIal": null, "sIze": "7801405440", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-sIze": "128", "type": "part", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": "/dev/sdd", "hctl": null, "tran": null, "subsystems": "block:scsi:usb:pci", "rev": null, "vendor": null, "zoned": "none"}
         ]
      },
      {"name": "/dev/sde", "kname": "/dev/sde", "maj:mIn": "8:64", "fstype": null, "mountpoInt": null, "label": null, "uuId": null, "parttype": null, "partlabel": null, "partuuId": null, "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": "WiFi Flash Disk ", "serIal": "4443A1825620F018", "sIze": "8022654976", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-sIze": "128", "type": "disk", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": null, "hctl": "10:0:0:0", "tran": "usb", "subsystems": "block:scsi:usb:pci", "rev": "1.00", "vendor": "Xiaomi  ", "zoned": "none",
         "children": [
            {"name": "/dev/sde1", "kname": "/dev/sde1", "maj:mIn": "8:65", "fstype": "exfat", "mountpoInt": null, "label": null, "uuId": "4F37-39EB", "parttype": "0x7", "partlabel": null, "partuuId": "3c76bd79-01", "partflags": null, "ra": "128", "ro": "0", "rm": "1", "hotplug": "1", "model": null, "serIal": null, "sIze": "8021606400", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alIgnment": "0", "mIn-Io": "512", "opt-Io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-sIze": "128", "type": "part", "dIsc-aln": "0", "dIsc-gran": "0", "dIsc-max": "0", "dIsc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": "/dev/sde", "hctl": null, "tran": null, "subsystems": "block:scsi:usb:pci", "rev": null, "vendor": null, "zoned": "none"}
         ]
      }
   ]
}
```

lurch, 2019-08-22:
@yagizhan49 and @queeup What language to you have your OS set to?

queeup, 2019-08-22:
```
$ locale
LANG=tr_TR.UTF-8
LANGUAGE=tr
LC_CTYPE="tr_TR.UTF-8"
LC_NUMERIC="tr_TR.UTF-8"
LC_TIME="tr_TR.UTF-8"
LC_COLLATE="tr_TR.UTF-8"
LC_MONETARY="tr_TR.UTF-8"
LC_MESSAGES="tr_TR.UTF-8"
LC_PAPER="tr_TR.UTF-8"
LC_NAME="tr_TR.UTF-8"
LC_ADDRESS="tr_TR.UTF-8"
LC_TELEPHONE="tr_TR.UTF-8"
LC_MEASUREMENT="tr_TR.UTF-8"
LC_IDENTIFICATION="tr_TR.UTF-8"
LC_ALL=
```

queeup, 2019-09-20:
1.5.57 also not recognize my usbs.

queeup, 2020-05-23:
`LC_ALL=C` or `LC_ALL="en_US.UTF-8"` solving the isue.

```console
LC_ALL=C balenaEtcher-1.5.91-x64.AppImage
```

thundron, 2020-05-29:
@queeup thank you. Maybe that could cause issue with translations in the future, so it's nice to know

evrifaessa, 2021-05-04:
A bit late, but my OS was also Turkish. In Turkish, the uppercase for letter "i" is not "I". It's "İ" (a special character in Turkish alphabet). That probably breaks the software.

lurch, 2021-05-04:
Ahhh, that sounds like a possibility! The column-names are specified in the `lsblk` sourcecode in uppercase https://github.com/karelzak/util-linux/blob/master/misc-utils/lsblk.c#L161 so there's probably some conversion-error with the letter `I` when it later converts the column-names to lower-case for printout? :man_shrugging:  I guess you could file a bug-report against `lsblk`, if you felt like it.

queeup, 2021-05-04:
Lets keep it open for a while. Probably this is out of scope but let's see some words from [util-linux](https://github.com/karelzak/util-linux) developers.

queeup, 2022-09-07:
Fixed util-linux > [v2.37-rc2](https://github.com/util-linux/util-linux/releases/tag/v2.37-rc2). Can be closed now. Thank you.

closed 2022-09-08 by mcraa