balena-io/etcher #2732: Stuck on initialising device for Compute Module 3 on macOS

dickolsson, 2019-04-14:
- **Etcher version:** 1.5.24
- **Operating system and architecture:** macOS Mojave
- **Image flashed:** 2019-04-08 Rasbian Stretch Lite
- **Do you see any meaningful error information in the DevTools?** No

When I try to flash a Raspberry Pi Compute Module 3 (not 3+) the initialisation process is getting stuck right at the end (see screenshot). I have tried this multiple times, both on a Macbook Air and a Macbook Pro (both Mojave).

The CM3 is mounted on the official CMV3 IO board, with the J4 USB SLAVE jumper in the ENABLE mode.

I have even tried the same process from a Debian Stretch desktop inside VMWare Fusion. Etcher perfectly detects the Compute Module when it's plugged in, but it's getting stuck at exactly the same spot as in the macOS screenshot below.

Any ideas of what's wrong here?

[image]

[image]

dickolsson, 2019-04-14:
Just noticed that the green 'ACT' light on the IO board turns off just as the progress bar reaches that last spot where it gets stuck. But I don't know what this means, or even if it's useful information...

thundron, 2019-04-30:
@dickolsson Got my hands on a CMV3 board right now, I'll let you know the results!

thundron, 2019-04-30:
@dickolsson What's the position of the J3 jumper though? I have it in `3V3` mode and it finished initializing without issues with a 16GB sdcard inserted (I got the CM3 without onboard storage)

dickolsson, 2019-05-01:
This issue can be closed. I had been sent a CM3 Lite in error, instead of a CM3 8GB, so there was no memory to flash to LOL :)