balena-io/etcher #2733: Lot of vulnerabilities and bad security

Jacalz, 2019-04-15:
- **Etcher version:** 1.5.24
- **Operating system and architecture:** Linux but I think this would be an issue on all operating systems.
- **Image flashed:** No image flushed, just building from source

## Description
Packages are slow to get updated and in general security vulnerabilities are left untreated for a long time. When running `npm audit` it reports 19 vulnerabilities. Where 17 are considered moderate and 2 high. You need to start paying closer attention to security issues and use `npm audit` and `npm audity fix` more often. This is exposing users machines and this has been the case for a long time.

I hope you take this seriously!

thundron, 2019-04-15:
We do, and we're renewing the codebase in smaller parts to start addressing minor issues as well as bigger ones. It's one of the reasons why we switched to a quicker release process, so that each fix comes individually in a new version, as opposed to having to wait for a release that includes other changes/fixes

lurch, 2019-04-15:
...although it's also worth bearing in mind that "traditionally" NodeJS apps run as server-side processes (i.e. directly exposed to the internet), whereas here they're being used in a locally-run electron app, so some of the security issues that affect server apps don't necessarily affect Etcher.

closed 2019-05-09 by zvin