balena-io/etcher #2735: `make electron-develop` errors???

AdamKearn, 2019-04-17:
- **Etcher version:** 
- **Operating system and architecture:**  Ubuntu 64bit

Hello, I am trying to compile etcher for windows.  I am building on Ubuntu though as I can't find any instructions for compiling on windows 10.

I have followed the guide and I am getting stuck.  Can anyone help me out?
I am not sure what is going wrong.

Thanks
Adam.

```
dev@ubuntu:~/etcher$ make electron-develop
./scripts/resin/electron/install.sh \
	-b /home/dev/etcher \
	-r x64 \
	-s linux \
	-m 6.7.0
node: v8.10.0
npm: 3.5.2
NPM configuration
; cli configs
long = true
user-agent = "npm/3.5.2 node/v8.10.0 linux x64"

; environment configs
arch = "x64"
build-from-source = true
disturl = "https://atom.io/download/electron"
fetch-retries = 20
fetch-retry-factor = 1.5
fetch-retry-maxtimeout = 10000
fetch-retry-mintimeout = 1000
loglevel = "warn"
progress = false
registry = "http://registry.npmjs.org/"
runtime = "electron"
spin = false
strict-ssl = false
target = "3.1.3"

; builtin config undefined
globalconfig = "/etc/npmrc"
globalignorefile = "/etc/npmignore"
prefix = "/usr/local"

; default values
access = null
also = null
always-auth = false
bin-links = true
browser = null
ca = null
cache = "/home/dev/.npm"
cache-lock-retries = 10
cache-lock-stale = 60000
cache-lock-wait = 10000
cache-max = null
cache-min = 10
cafile = undefined
cert = null
color = true
depth = null
description = true
dev = false
dry-run = false
editor = "vi"
engine-strict = false
; fetch-retries = 2 (overridden)
; fetch-retry-factor = 10 (overridden)
; fetch-retry-maxtimeout = 60000 (overridden)
; fetch-retry-mintimeout = 10000 (overridden)
force = false
git = "git"
git-tag-version = true
global = false
global-style = false
; globalconfig = "/usr/local/etc/npmrc" (overridden)
; globalignorefile = "/usr/local/etc/npmignore" (overridden)
group = 1000
heading = "npm"
https-proxy = null
if-present = false
ignore-scripts = false
init-author-email = ""
init-author-name = ""
init-author-url = ""
init-license = "ISC"
init-module = "/home/dev/.npm-init.js"
init-version = "1.0.0"
json = false
key = null
legacy-bundling = false
link = false
local-address = undefined
loglevel = "warn"
; long = false (overridden)
message = "%s"
node-version = "8.10.0"
npat = false
onload-script = null
only = null
optional = true
parseable = false
; prefix = "/usr" (overridden)
production = false
; progress = true (overridden)
proprietary-attribs = true
proxy = null
rebuild-bundle = true
; registry = "https://registry.npmjs.org/" (overridden)
rollback = true
save = false
save-bundle = false
save-dev = false
save-exact = false
save-optional = false
save-prefix = "^"
scope = ""
searchexclude = null
searchopts = ""
searchsort = "name"
shell = "/bin/bash"
shrinkwrap = true
sign-git-tag = false
; strict-ssl = true (overridden)
tag = "latest"
tag-version-prefix = "v"
tmp = "/tmp"
umask = 18
unicode = true
unsafe-perm = true
usage = false
user = 0
; user-agent = "npm/{npm-version} node/{node-version} {platform} {arch}" (overridden)
userconfig = "/home/dev/.npmrc"
version = false
versions = false
viewer = "man"

Installing dependencies
npx: installed 426 in 10.601s

> balena-etcher@1.5.27 install /home/dev/etcher
> node-gyp rebuild

make[1]: Entering directory '/home/dev/etcher/build'
  SOLINK_MODULE(target) Release/obj.target/elevator.node
  COPY Release/elevator.node
make[1]: Leaving directory '/home/dev/etcher/build'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: winusb-driver-generator@1.2.4 (node_modules/winusb-driver-generator):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for winusb-driver-generator@1.2.4: wanted {"os":"win32","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.8 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.8: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

audited 15850 packages in 15.007s
found 73 vulnerabilities (6 low, 39 moderate, 28 high)
  run `npm audit fix` to fix them, or `npm audit` for details
Applying /home/dev/etcher/patches/allow-electron-forks-of-modules-that-use-pre-gyp.patch to /home/dev/etcher
patching file node_modules/node-pre-gyp/lib/util/versioning.js
Hunk #1 FAILED at 80.
Hunk #2 FAILED at 263.
2 out of 2 hunks FAILED -- saving rejects to file node_modules/node-pre-gyp/lib/util/versioning.js.rej
Makefile:97: recipe for target 'electron-develop' failed
make: *** [electron-develop] Error 1
dev@ubuntu:~/etcher$ 
```

thundron, 2019-04-18:
@AdamKearn Try running that command with an additional env variable `NPM_VERSION` set to `6.7.0`
Let us know how it goes!

thundron, 2019-04-18:
Also fwiw: we recently made some changes to how we build packages and were reiterating over it many times until we found an ideal solution, so the documentation is going to be updated soon with the new steps

AdamKearn, 2019-04-20:
Hello,  Sorry for the late reply I have been soo busy with revising for my exams :(
I have done some debugging and I still can't figure what is going wrong...  I did try what you suggested:
`NPM_VERSION 6.7.0` and also `NPM_VERSION 6.9.0`  but they both returned with the same result.

I was getting a lot of errors saying "permission denied" so I changed to the `root` user, and those went away.

Do you think it would be best for me to wait until the documentation has been updated and come back to this project afterwards?  Say, next month or will it take longer? 

Sorry for quite a long response btw.

-Adam

```
root@ubuntu:/home/dev/etcher# node -v
v6.7.0
root@ubuntu:/home/dev/etcher# npm -v
6.9.0
root@ubuntu:/home/dev/etcher#
```

```
root@ubuntu:/home/dev/etcher# make electron-develop NPM_VERSION=6.9.0
./scripts/resin/electron/install.sh \
	-b /home/dev/etcher \
	-r x64 \
	-s linux \
	-m 6.9.0
node: v6.7.0
npm: 6.9.0
NPM configuration
; cli configs
long = true
metrics-registry = "http://registry.npmjs.org/"
scope = ""
user-agent = "npm/6.9.0 node/v6.7.0 linux x64"

; environment configs
arch = "x64"
build-from-source = true
disturl = "https://atom.io/download/electron"
fetch-retries = 20
fetch-retry-factor = 1.5
fetch-retry-maxtimeout = 10000
fetch-retry-mintimeout = 1000
loglevel = "warn"
progress = false
registry = "http://registry.npmjs.org/"
runtime = "electron"
spin = false
strict-ssl = false
target = "3.1.3"

; default values
access = null
allow-same-version = false
also = null
always-auth = false
audit = true
audit-level = "low"
auth-type = "legacy"
before = null
bin-links = true
browser = null
ca = null
cache = "/root/.npm"
cache-lock-retries = 10
cache-lock-stale = 60000
cache-lock-wait = 10000
cache-max = null
cache-min = 10
[... 127 lines cut ...]
root@ubuntu:/home/dev/etcher# npm audit fix
updated 1 package in 14.982s
fixed 7 of 24 vulnerabilities in 15540 scanned packages
  17 vulnerabilities required manual review and could not be updated
root@ubuntu:/home/dev/etcher#
```

rerunning `make electron-develop`

```
Installing dependencies
npx: installed 426 in 12.458s
npm WARN lifecycle balena-etcher@1.5.27~install: cannot run in wd balena-etcher@1.5.27 node-gyp rebuild (wd=/home/dev/etcher)
updated 1 package and audited 15540 packages in 17.669s
found 24 vulnerabilities (2 low, 18 moderate, 4 high)
  run `npm audit fix` to fix them, or `npm audit` for details
Applying /home/dev/etcher/patches/allow-electron-forks-of-modules-that-use-pre-gyp.patch to /home/dev/etcher
patching file node_modules/node-pre-gyp/lib/util/versioning.js
Hunk #1 FAILED at 80.
Hunk #2 FAILED at 263.
2 out of 2 hunks FAILED -- saving rejects to file node_modules/node-pre-gyp/lib/util/versioning.js.rej
Makefile:97: recipe for target 'electron-develop' failed
make: *** [electron-develop] Error 1
root@ubuntu:/home/dev/etcher# 
```

AdamKearn, 2019-04-20:
Also, should I be working off the `master` branch or is there a more up to date release I should try?

thundron, 2019-05-14:
Solved with 792fab20e68a3428dbf1c6d1052c9ebd472b05f6. Closing

closed 2019-05-14 by thundron