balena-io/etcher #2737: image etched to 64gb sd card, will not boot.. same image to 32gb card works okl

sdetweil, 2019-04-17:
- **Etcher version:** 
1.5.4
- **Operating system and architecture:** 
WIndows 10 64bit
- **Image flashed:** 
nvidia jetson nano
https://developer.nvidia.com/embedded/dlc/jetson-nano-dev-kit-sd-card-image
- **Do you see any meaningful error information in the DevTools?** 
no.. says completed ok

sdetweil, 2019-04-17:
image is ubuntu 18.04

thundron, 2019-04-18:
@sdetweil Probably due to this https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable , as mentioned there Etcher doesn't modify the image in any way; as a result, some images need some additional steps
Also look into this for more infos on bootable ubuntu isos: https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

lurch, 2019-04-23:
If Etcher says the writing completed okay, it sounds to me like it might be a problem with the "nvidia jetson nano" rather than a problem with Etcher? Perhaps it's not compatible with such large SD cards? :man_shrugging:

szb512, 2019-05-03:
Have you tried reflashing the SD card with the image?

sdetweil, 2019-05-03:
Tried 3 times. Would not boot all three. Used a 32gig card, booted 1st time

szb512, 2019-05-04:
Does the .zip file have a .iso (disk image file)?

sdetweil, 2019-05-04:
not iso, but img

jetson-nano-sd-r32.1-2019-03-18.img

you can download it from the link in the initial post

szb512, 2019-05-04:
Have you tried redownloading the file?

sdetweil, 2019-05-04:
yes, i have it without problem... unzipped and then used etcher to install the .img to the card...

32gig card boots after, 64gig card does not

szb512, 2019-05-04:
It's probably a problem with the file, I guess it can't support 64 GB cards.

talkingtab, 2019-05-09:
I have a jetson nano, used etcher on my mac to flash the image onto a 64G card and it was able to boot just fine.  

One thing is that I did not unzip the nano image file, just selected the zip file with etcher.

sdetweil, 2019-05-09:
@talkingtab  thank you.. that worked!

closed 2019-05-09 by thundron

lurch, 2019-05-10:
How odd! :confused:

sdetweil, 2019-05-10:
@lurch exactly!

lurch, 2019-05-10:
@zvin or @thundron Any ideas what might explain this unusual behaviour? (just out of curiosity...)

thundron, 2019-05-10:
Well tbh we fixed many things since v1.5.4 so you could maybe try flashing the extracted image with the recent `v1.5.33` "for science" @sdetweil @talkingtab 
I'll try to get my hands on a jetson nano to check it but this will require some more time

talkingtab, 2019-05-10:
@thundron I will try to  run some experiments on my system. It will probably take a couple of days, but the variables I can think of are:

- macos vs windows
- v1.5.4 vs v1.5.33 vs v1.4.9
- nano vs everything else

The first two seem much more likely to me than it being a nano issue.  
My current version on the mac reports: balenaEtcher, v1.4.9, so I guess the first step is try the most recent versions.

sdetweil, 2019-05-10:
i am using Windows 10, and etcher 1.5.24
was just offered 1.5.33 as first update

lurch, 2019-05-10:
> is the best way to get v1.5.4. for the mac to clone the repo && build it myself?

Old releases can be found at https://github.com/balena-io/etcher/releases

> can I use etcher to go from SD => img ?

Not yet :wink: #266

talkingtab, 2019-05-12:
I tried this on a mac on with 1.5.4 and did not have a problem. I was offered 1.5.33 as an upgrade but when I clicked on the "download" button it took me to a page that only had 1.5.4. Not sure it is worth pursuing this, at least on the mac.

lurch, 2019-05-12:
> I was offered 1.5.33 as an upgrade but when I clicked on the "download" button it took me to a page that only had 1.5.4

See #2773

talkingtab, 2019-05-13:
Not sure I got the release info right in some of the earlier messages. In any case I have 1.4.9, 1.5.24 and 1.5.35. I tested the nano file extracted from the zip to an .img and successfully booted the SD card on all versions.  This is on a Mac running 10.14.1
Thanks @lurch for the help getting 1.5.35.