balena-io/etcher #2739: Failed to fetch https://deb.etcher.io/dists/stable/InRelease

Bicash, 2019-04-18:
- **Etcher version:** None
- **Operating system and architecture:** Ubuntu 18.04
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** No

I`m running this commands for install etcher:
```
echo "deb https://deb.etcher.io stable etcher" | sudo tee /etc/apt/sources.list.d/balena-etcher.list
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
sudo apt-get update
```
On update command this error shown:
```
Err:18 https://deb.etcher.io stable InRelease                                  
  Could not connect to deb.etcher.io:443 (104.27.179.91), connection timed out Could not connect to deb.etcher.io:443 (104.27.178.91), connection timed out
Fetched 252 kB in 46s (5 532 B/s)               
Reading package lists... Done
W: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  Could not connect to deb.etcher.io:443 (104.27.179.91), connection timed out Could not connect to deb.etcher.io:443 (104.27.178.91), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Please help me install Etcher.

Dragon8oy, 2019-04-19:
I have the same issue on Ubuntu 18.04, Ubuntu 18.10 and Ubuntu 19.04.
`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61` returns
`gpg: keyserver receive failed: Server indicated a failure` on all of the versions.

I've only tested those 3 because I was updating 3 machines

lurch, 2019-04-23:
Duplicate of #2536 ?

closed 2019-12-30 by thundron

AndreiCherniaev, 2021-04-12:
The same problem on my Ubuntu 20.10 x86_64 while updating:
_Err:9 https://dl.bintray.com/etcher/debian stable InRelease         
  403  Forbidden [IP: 35.156.125.116 443]
Reading package lists... Done                    
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 35.156.125.116 443]
E: The repository 'https://deb.etcher.io stable InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details._

graugelb, 2021-05-03:
Selbes Problem mit Mint LMDE4 - >
"Fehlschlag beim Holen von https://deb.etcher.io/dists/stable/InRelease 403  Forbidden [IP: 3.124.41.190 443]Das Depot »https://deb.etcher.io stable InRelease« ist nicht signiert."

--> Duplicate of #2536 Nein. Auch  weiter: Sudo apt autoremove --> negativ.
--> apt-transport-https ist schon die neueste Version (1.8.2.3).
Negative:  SebastianCaballero commented on 25 Dec 2020
The solution if you want to remove Etcher (the package) in order to generate the update of the packages is:
sudo nano /etc/apt/sources.list
remove those lines that contains the hhtp:// domain of the problem. Then CTRL+X, yes and ENTER.
Now you can sudo apt update.
Nein.-->
E: Fehlschlag beim Holen von https://deb.etcher.io/dists/stable/InRelease 403  Forbidden [IP: 3.125.168.216 443]
E: Das Depot »https://deb.etcher.io stable InRelease« ist nicht signiert.
N: Eine Aktualisierung von solch einem Depot kann nicht auf eine sichere Art durchgeführt werden, daher ist es standardmäßig deaktiviert.
N: Weitere Details zur Erzeugung von Paketdepots sowie zu deren Benutzerkonfiguration finden Sie in der Handbuchseite apt-secure(8).
Hmmmm ?

lurch, 2021-05-03:
See #3501

krishnaTORQUE, 2021-05-03:
I am facing the same problem 

Error
```
Err:3 https://dl.bintray.com/etcher/debian stable InRelease                                                                                                                        
  403  Forbidden [IP: 34.214.135.19 443]
Reading package lists... Done                                                                                                                                                      
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 34.214.135.19 443]
E: The repository 'https://deb.etcher.io stable InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

graugelb, 2021-05-07:
> See #3501

It worked. But only almost..?
During the procedure #3501 (I had to install first the "curl" command) I got at the end:.
.
.
.
.Vormals nicht ausgewähltes Paket balena-etcher-electron wird gewählt.
Vorbereitung zum Entpacken von .../balena-etcher-electron_1.5.117_amd64.deb ...
Entpacken von balena-etcher-electron (1.5.117) ...
libgconf-2-4:amd64 (3.2.6-5) wird eingerichtet ...
gconf-service (3.2.6-5) wird eingerichtet ...
gconf2 (3.2.6-5) wird eingerichtet ...
balena-etcher-electron (1.5.117) wird eingerichtet ...

> chmod: Zugriff auf '/opt/balena-etcher-electron/chrome-sandbox' nicht möglich: Datei oder Verzeichnis nicht gefunden

Trigger für desktop-file-utils (0.23-4) werden verarbeitet ...
Trigger für mime-support (3.62) werden verarbeitet ...
Trigger für hicolor-icon-theme (0.17-2) werden verarbeitet ...
Trigger für gnome-menus (3.31.4-3) werden verarbeitet ...
Trigger für libc-bin (2.28-10) werden verarbeitet ...
Trigger für man-db (2.8.5-2) werden verarbeitet ..

So I am not sure at the end if this is essential?? (I am a what we call in German "DAU") 

Question: 
Is the task executed??
Pls. yes/no ?
Regards