balena-io/etcher #2742: Pre-select image from command line

dmke, 2019-04-18:
- **Etcher version:** 1.5.24, installed via `.deb` package from Github releases
- **Operating system and architecture:** Debian 9, amd64
- **Image flashed:** custom buildroot image for Raspberry
- **Do you see any meaningful error information in the DevTools?** irrelevant

Is it possible to pass the path to an image on the command line? This could boost my productivity a lot :-)

Just running

```console
$ balena-etcher-electron /path/to/buildroot/output/images/sdcard.img
```

didn't yield the expected result; the GUI still asked me to select an image.

(If you don't plan on implementing this yourselves: will you accept a PR implementing this?)

<details>
<summary>Side note</summary>

I also expected

```console
$ balena-etcher-electron --help
```

to print something meaningful. Instead the GUI just popped up.

</details>

thundron, 2019-04-22:
I don't think we recently discussed this, let us check and we'll get back to you

lurch, 2019-04-23:
Previously requested in #1097

dmke, 2019-04-24:
Looks like there was a CLI, but that was removed without much explanation in #2630 (v1.5.0).

Are there plans to revive this again or integrate this into the GUI app?

thundron, 2019-04-25:
@dmke It was removed due to it not being used as much as we thought as opposed to having an entirely separate package building process, that also needs to be maintained.
Whenever you feel like it you can issue a PR with this feature, it sounds like a nice first step into having a headless Etcher!

thundron, 2020-05-29:
Should (finnally!) be possible with https://github.com/balena-io/etcher/pull/3177