balena-io/etcher #2743: "Error Opening image" Something went wrong while opening jetson-nano-sd-r32.1-2019-03-18.zip  Error: end of central directory record signiture not found.

bw1995, 2019-04-18:
- **Etcher version:** 1.5.24-x64
- **Operating system and architecture:** Win10Pro 64bit
- **Image flashed:**  jetson-nano-sd-r32.1-2019-03-18.zip
[image]

- **Do you see any meaningful error information in the DevTools?** No

lurch, 2019-04-23:
Are you sure the .zip file you've chosen has been downloaded correctly? Are other ZIP tools able to open it okay?

szb512, 2019-04-23:
Have you tried 7zip?

bw1995, 2019-04-25:
The downloaded file was corrupt. Took me five attempts to download a good file.  Then etcher worked fine.
Thanks for the follow-up.
Please close trouble ticket.
Bill

From: szb512
Sent: Tuesday, April 23, 2019 6:36 PM
To: balena-io/etcher
Cc: Bill; Author
Subject: Re: [balena-io/etcher] "Error Opening image" Something went wrongwhile opening jetson-nano-sd-r32.1-2019-03-18.zip Error: end of centraldirectory record signiture not found. (#2743)

Have you tried 7zip?

closed 2019-04-29 by thundron