balena-io/etcher #2745: x64 AppImage build is missing in 1.5.28 release

hgmtnd, 2019-04-19:
- **Etcher version:** 1.5.28
- **Operating system and architecture:** Linux x64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

Release 1.5.28 is missing the x64 AppImage package.
https://github.com/balena-io/etcher/releases/tag/v1.5.28 has only 23 assets instead of 24.

```console
$ curl -I 'https://github.com/balena-io/etcher/releases/download/v1.5.28/balenaEtcher-1.5.28-ia32.AppImage'
HTTP/1.1 302 Found

$ curl -I 'https://github.com/balena-io/etcher/releases/download/v1.5.26/balenaEtcher-1.5.28-x64.AppImage'
HTTP/1.1 404 Not Found
```

thundron, 2019-04-22:
@hgmtnd Seems like you were searching for a 1.5.28 file on a `1.5.26` release, check the version after `/releases/...`

hgmtnd, 2019-04-22:
@thundron Good catch, that was mistake I made when preparing the report to post here, but the correct URL also results in 404:

```console
$ wget 'https://github.com/balena-io/etcher/releases/download/v1.5.28/balenaEtcher-1.5.28-x64.AppImage'
--2019-04-22 09:20:48--  https://github.com/balena-io/etcher/releases/download/v1.5.28/balenaEtcher-1.5.28-x64.AppImage
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/45055693/d6d9a800-6213-11e9-863b-acd2d07a399b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190422%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190422T122049Z&X-Amz-Expires=300&X-Amz-Signature=6bea6206fc2e572f0694af0382c0626a96ba3c36810734640fc58a3fe9439007&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DbalenaEtcher-1.5.28-x64.AppImage&response-content-type=application%2Foctet-stream [following]
--2019-04-22 09:20:49--  https://github-production-release-asset-2e65be.s3.amazonaws.com/45055693/d6d9a800-6213-11e9-863b-acd2d07a399b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190422%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190422T122049Z&X-Amz-Expires=300&X-Amz-Signature=6bea6206fc2e572f0694af0382c0626a96ba3c36810734640fc58a3fe9439007&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DbalenaEtcher-1.5.28-x64.AppImage&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.114.3
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.114.3|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-04-22 09:20:50 ERROR 404: Not Found.

$ wget 'https://github.com/balena-io/etcher/releases/download/v1.5.28/balenaEtcher-1.5.28-ia32.AppImage'
--2019-04-22 09:21:05--  https://github.com/balena-io/etcher/releases/download/v1.5.28/balenaEtcher-1.5.28-ia32.AppImage   
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/45055693/de16ab00-623f-11e9-8fa4-b3c223ff879a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190422%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190422T123706Z&X-Amz-Expires=300&X-Amz-Signature=23cd7eec91b516f617a002d3e0114d767c4d6a2c6bd2e8fd656385141bc71a13&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DbalenaEtcher-1.5.28-ia32.AppImage&response-content-type=application%2Foctet-stream [following]
--2019-04-22 09:21:06--  https://github-production-release-asset-2e65be.s3.amazonaws.com/45055693/de16ab00-623f-11e9-8fa4-b3c223ff879a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190422%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190422T123706Z&X-Amz-Expires=300&X-Amz-Signature=23cd7eec91b516f617a002d3e0114d767c4d6a2c6bd2e8fd656385141bc71a13&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DbalenaEtcher-1.5.28-ia32.AppImage&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.136.115
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.136.115|:443... connected.
HTTP request sent, awaiting response... 200 OK
```

As I mentioned previously, version 1.5.28 only has 23 assets, while all others have 24.

Anyway, v1.5.29 has already been released and it has all 24 assets, so this issue is obsolete now. Thank you.

[image]

[image]

closed 2019-04-22 by hgmtnd