balena-io/etcher #2746: BalenaEcher icon disappears from Ubuntu home menu after some little time.

Nayananga, 2019-04-19:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Hi, I use balenaEcher regularly but when I extract the zip and install the app image it appears in my ubuntu home menu, so I can reach it whenever I want easily. But after some little time when I search in my menu for the application, it is no longer there so I have to do the same process again and again.

lurch, 2019-04-23:
The AppImage doesn't get "installed", it simply gets extracted somewhere and run in-place (from wherever you extracted the zip to). So I suspect the "Ubuntu home menu" is probably just a cache of the most-recently-launched programs, and after "some little time" the Etcher AppImage expires from the cache.

If you want the balenaEtcher app to **always** be in your home menu, I think installing the .deb version as detailed [here](https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64) should work?

Roesjka, 2019-04-30:
The .deb repository only shows version 1.5.19. So it is not uptodate with the later versions.
Cheers.

lurch, 2019-04-30:
@Roesjka They release updates to the .deb and .rpm repos at a much slower rate than they release to their website, see e.g. #2657
I _think_ the theory is that this keeps the repo-versions more stable than the website-versions :man_shrugging:  And 1.5.19 was only released a month ago anyway... https://github.com/balena-io/etcher/blob/master/CHANGELOG.md#v1519

Roesjka, 2019-04-30:
I was aware it was only a month ago ;), was just wondering why it wasn't in sync with the rest, thanks for the explanation.

closed 2023-02-25 by Nayananga