balena-io/etcher #2747: Etcher is corrupting USB drive

godfrey62, 2019-04-20:
- **Etcher version:**  1.5.24
- **Operating system and architecture:** macOS Mojave 10.14.4, MacBookPro 15" Retina 2014
- **Image flashed:** ubuntu-18.04.2-desktop-amd64.iso
- **Do you see any meaningful error information in the DevTools?** No

I'm trying to use Etcher to create a bootable USB drive to install Ubuntu 18.04.2 on an old MacBook.

The USG is formatted to FAT32, petition type GUID. After formatting the USB drive is mounted on macOS Mojave MacBook.

I then run Etcher to flash the ISO onto the mounted USB drive. After flashing the verify process reports that all is well, but immediately macOS Mojave throws up an error message that "The disk you inserted was not readable by this computer".

Terminal diskutil reports:

/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     Apple_partition_scheme                        *15.5 GB    disk2
   1:        Apple_partition_map                         4.1 KB     disk2s1
   2:                  Apple_HFS                         2.5 MB     disk2s2

for the unmounted volume.

The USB is not "destroyed" in the sense that I can reformat it to FAT32 GUID and it mounts ok once again. It's just every time I use Etcher the result is an unreadable USB drive (never mind whether bootable or not).

Judging by the volume of closed posts on this issue (none of which seem to have been resolved) this is not an uncommon problem.

Any suggestions on how to resolve this issue would be appreciated.

thundron, 2019-04-22:
This is not an issue with Etcher but rather common knowledge that often seems to be missing: 
First of all, etcher doesn't apply any logic when flashing so whatever is in the image, *completely* overwrites what's in the target that is flashed; as a result from that, some OSes (mainly Windows and OSX afaik) might not be able to read all (or none at all) of the partitions that ***your** image* you just flashed contained. As a test, you can try connecting your target on a Linux machine to see if it recognizes *some* partitions, which again is not something **we** control but the users *should* be aware of, as they're flashing an image that completely erases all data from the target (USB, sdcard or device) and can make it unreadable by *some* OSes (again, like Windows and OSX).
If you want to recover your target and use it for something else, you can try one of the suggested steps in our [user docs](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives)

closed 2019-04-22 by thundron

en-gee77, 2020-02-09:
You just cba, laughable