balena-io/etcher #2748: error downloading packages

ronogio, 2019-04-22:
- **Etcher version:** 
- **Patchbox OS, armv7l:** 

Hey to everyone!
I'm totally new to linux, so sorry if I will not comprehend everything or if I am asking something stupid.-
I'm trying to install etcher following [instructions here](https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64) bu after the first commands:

`echo "deb https://deb.etcher.io stable etcher" | sudo tee /etc/apt/sources.list.d/balena-etcher.list`

`patch@patchbox:~ $ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.EiY5NdwTea/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: key 379CE192D401AB61: "Bintray (by JFrog) <bintray@bintray.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
`
I get this message:

`patch@patchbox:~ $ sudo apt-get update
Trovato:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease           
Trovato:2 http://archive.raspberrypi.org/debian stretch InRelease              
Trovato:3 http://blokas.io/apt rpi InRelease                                   
Ign:4 https://deb.etcher.io stable InRelease          
Scaricamento di:5 https://dl.bintray.com/etcher/debian stable Release [3.674 B]
Trovato:5 https://dl.bintray.com/etcher/debian stable Release
Lettura elenco dei pacchetti... Fatto
N: Skipping acquire of configured file 'etcher/binary-armhf/Packages' as repository 'https://deb.etcher.io stable InRelease' doesn't support architecture 'armhf`

So I dont know how to continue since when I try to run `sudo apt-get install balena-etcher-electron` it says that it is impossible to find the package!

Thanks a lot

lurch, 2019-04-23:
Etcher only provides packages for x86 and x64, AFAIK they don't provide packages for armhf. See e.g. #1992

ronogio, 2019-04-23:
what a pity!
thanks anyway

thundron, 2019-04-29:
@ronogio as @lurch kindly pointed out, as of now we only distribute packages for x86 and x64. Closing

closed 2019-04-29 by thundron

lurch, 2019-11-03:
Progress being tracked in #2870