balena-io/etcher #2750: Allow user to enter wifi credentials in GUI, save to wpa_supplicant.conf, and store in boot partition

KM4YRI, 2019-04-23:
This is a feature request, my apologies if it's already been addressed, I wasn't able to find anything while searching through previous issues.

Raspbian and other Raspberry Pi distros have a feature to read an optional file on the boot partition named `wpa_supplicant.conf`.  If found during boot, it will be copied to the system.  This is extremely useful because it allows headless installs.  No need to connect a monitor, just drop the file on and ssh into the Pi when it comes up.

It think it would be extremely useful if Etcher had the ability to optionally prompt the user for their username and password.  These credentials would be written to a file and stored in the boot partition, after the image has been copied and verified.

The order of operations would be as follows:

1. Open the etcher app
2. Select the file
3. Select the drive
4. Flash the image
5. Verify the image
6. Detect if a 'boot' partition is present.  Determine if it appears to be a Raspberry Pi boot partition by looking at while files are present.
7. If it is a Raspberry Pi boot partition, prompt the user if they would like to enter wifi credentials, with re-assurance they they will only be saved to the Pi's boot partition.
8. If user selects, Yes, create the file and write it to the boot partition
9. Eject / unmount the drive, if the option to do so is selected.

A boolean checkbox should be added to the Settings page, "Prompt and store wifi credentials in boot partition".  It should be unchecked by default, as to not derail the user experience from what most normal users have dealt with.  This would allow power users and those that would gain value from it to enable it when needed.

Thoughts?

lurch, 2019-04-23:
Sounds like a duplicate of #718 ?

KM4YRI, 2019-04-29:
@lurch I think you're correct.  Feel free to mark this issue as such.