balena-io/etcher #2751: gpg: keyserver receive failed: Server indicated a failure

Dragon8oy, 2019-04-23:
- **Etcher version:** Attempted to install latest
- **Operating system and architecture:**  Ubuntu 19.04 x86_64

Upon running `sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61`

It returns: `Executing: /tmp/apt-key-gpghome.hAVPBRTG2F/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: Server indicated a failure`

Then when attempting to run `sudo apt update` it returns:
`GPG error: https://dl.bintray.com/etcher/debian stable Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 379CE192D401AB61`

lurch, 2019-04-23:
Duplicate of #2612 ?

Dragon8oy, 2019-04-23:
Yeah, sorry. No fix yet though but still a duplicate.

closed 2019-04-23 by Dragon8oy