balena-io/etcher #2752: undefined% Validating

pattyland, 2019-04-24:
- **Etcher version:** 1.5.19
- **Operating system and architecture:** Windows 10 64-Bit
- **Image flashed:** https://github.com/home-assistant/hassos/releases/download/2.11/hassos_intel-nuc-2.11.img.gz

- **Do you see any meaningful error information in the DevTools?** 
[image]

lurch, 2019-04-24:
Does the same thing still happen with the latest version of Etcher? https://www.balena.io/etcher/#download

pattyland, 2019-04-24:
I tried version 1.5.30 and the display of percent for flashing is wrong too:
```2C:\Users\somuelle\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Apr 24 2019 16:44:27 GMT+0200 (Mitteleuropäische Sommerzeit) Flashing 1 devices, undefined% at 4.8 MB/s (total 4.8 MB/s) eta in undefineds with 0 failed devices
2C:\Users\somuelle\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Apr 24 2019 16:44:28 GMT+0200 (Mitteleuropäische Sommerzeit) Flashing 1 devices, undefined% at 4.8 MB/s (total 4.8 MB/s) eta in undefineds with 0 failed devices
2C:\Users\somuelle\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Apr 24 2019 16:44:29 GMT+0200 (Mitteleuropäische Sommerzeit) Flashing 1 devices, undefined% at 4.8 MB/s (total 4.8 MB/s) eta in undefineds with 0 failed devices
````

Josephvb10, 2019-04-30:
Same problem on the latest Linux build.

zvin, 2019-05-06:
Fixed in 1.5.33

closed 2019-05-06 by zvin