balena-io/etcher #2754: regression: writing image with bmap results in corrupt data

dlech, 2019-04-26:
- **Etcher version:** 1.5.30
- **Operating system and architecture:** Win 10 1809 x64
- **Image flashed:** https://education.lego.com/en-us/support/mindstorms-ev3/python-for-ev3
- **Do you see any meaningful error information in the DevTools?** No

This image uses a bmap to speed up flash time.

Flashing the same image with Etcher 1.4.9 does not corrupt the image.

dlech, 2019-04-26:
Here is a binary diff showing where a file (`uEnv.txt` in the FAT partition) is written to a different location from where it should be.

<pre><span style="background-color:#839496"><font color="#002B36">   Visuel HexDiff v 0.0.53 by tTh 2007                             dec   7bits  </font></span>
 4538416   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 4538432   00 00 <span style="background-color:#839496"><font color="#002B36">03</font></span> <span style="background-color:#839496"><font color="#002B36">18</font></span> 00 00 00 00 <span style="background-color:#839496"><font color="#002B36">23</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">42</font></span> <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">74</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">73</font></span>      <span style="background-color:#839496"><font color="#002B36">  </font></span>    <span style="background-color:#839496"><font color="#002B36"># Boot s</font></span>
 4538448   <span style="background-color:#839496"><font color="#002B36">63</font></span> <span style="background-color:#839496"><font color="#002B36">72</font></span> <span style="background-color:#839496"><font color="#002B36">69</font></span> <span style="background-color:#839496"><font color="#002B36">70</font></span> <span style="background-color:#839496"><font color="#002B36">74</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">74</font></span> <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">63</font></span> <span style="background-color:#839496"><font color="#002B36">68</font></span> <span style="background-color:#839496"><font color="#002B36">61</font></span> <span style="background-color:#839496"><font color="#002B36">69</font></span> <span style="background-color:#839496"><font color="#002B36">6e</font></span> <span style="background-color:#839496"><font color="#002B36">2d</font></span> <span style="background-color:#839496"><font color="#002B36">6c</font></span>    <span style="background-color:#839496"><font color="#002B36">cript to chain-l</font></span>
 4538464   <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">61</font></span> <span style="background-color:#839496"><font color="#002B36">64</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">6e</font></span> <span style="background-color:#839496"><font color="#002B36">65</font></span> <span style="background-color:#839496"><font color="#002B36">77</font></span> <span style="background-color:#839496"><font color="#002B36">65</font></span> <span style="background-color:#839496"><font color="#002B36">72</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">55</font></span> <span style="background-color:#839496"><font color="#002B36">2d</font></span> <span style="background-color:#839496"><font color="#002B36">42</font></span> <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">74</font></span>    <span style="background-color:#839496"><font color="#002B36">oad newer U-Boot</font></span>
 4538480   <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">66</font></span> <span style="background-color:#839496"><font color="#002B36">72</font></span> <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">6d</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">6f</font></span> <span style="background-color:#839496"><font color="#002B36">66</font></span> <span style="background-color:#839496"><font color="#002B36">66</font></span> <span style="background-color:#839496"><font color="#002B36">69</font></span> <span style="background-color:#839496"><font color="#002B36">63</font></span> <span style="background-color:#839496"><font color="#002B36">69</font></span> <span style="background-color:#839496"><font color="#002B36">61</font></span> <span style="background-color:#839496"><font color="#002B36">6c</font></span> <span style="background-color:#839496"><font color="#002B36">20</font></span> <span style="background-color:#839496"><font color="#002B36">4c</font></span>    <span style="background-color:#839496"><font color="#002B36"> from official L</font></span>
[... cut ...]
 4539008   45 47 4f 20 66 69 72 6d 77 61 72 65 0a 23 0a 23    EGO firmware # #
 4539024   20 54 68 65 20 55 2d 42 6f 6f 74 20 74 68 61 74     The U-Boot that
 4539040   20 73 68 69 70 73 20 77 69 74 68 20 74 68 65 20     ships with the
 4539056   6f 66 66 69 63 69 61 6c 20 66 69 72 6d 77 61 72    official firmwar
 4539072   65 20 66 6f 72 20 4c 45 47 4f 20 4d 49 4e 44 53    e for LEGO MINDS
<span style="background-color:#839496"><font color="#002B36">   /media/VMs/bad-ev3dev.img                        15931539456  4538928   0%   </font></span>

</pre>

dlech, 2019-04-26:
The same image flashed with v1.4.9 has the file in the right place:

<pre><span style="background-color:#839496"><font color="#002B36">   Visuel HexDiff v 0.0.53 by tTh 2007                             dec   7bits  </font></span>
 4538416   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 4538432   00 00 03 18 00 00 00 00 23 20 42 6f 6f 74 20 73            # Boot s
 4538448   63 72 69 70 74 20 74 6f 20 63 68 61 69 6e 2d 6c    cript to chain-l
 4538464   6f 61 64 20 6e 65 77 65 72 20 55 2d 42 6f 6f 74    oad newer U-Boot
 4538480   20 66 72 6f 6d 20 6f 66 66 69 63 69 61 6c 20 4c     from official L
 4538496   45 47 4f 20 66 69 72 6d 77 61 72 65 0a 23 0a 23    EGO firmware # #
 4538512   20 54 68 65 20 55 2d 42 6f 6f 74 20 74 68 61 74     The U-Boot that
 4538528   20 73 68 69 70 73 20 77 69 74 68 20 74 68 65 20     ships with the
 4538544   6f 66 66 69 63 69 61 6c 20 66 69 72 6d 77 61 72    official firmwar
 4538560   65 20 66 6f 72 20 4c 45 47 4f 20 4d 49 4e 44 53    e for LEGO MINDS
<span style="background-color:#839496"><font color="#002B36">** /media/VMs/ev3dev-stretch-ev3-generic-2019-03-03.1887436800  4538416   0%    </font></span>
 4538416   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 4538432   00 00 03 18 00 00 00 00 23 20 42 6f 6f 74 20 73            # Boot s
 4538448   63 72 69 70 74 20 74 6f 20 63 68 61 69 6e 2d 6c    cript to chain-l
 4538464   6f 61 64 20 6e 65 77 65 72 20 55 2d 42 6f 6f 74    oad newer U-Boot
 4538480   20 66 72 6f 6d 20 6f 66 66 69 63 69 61 6c 20 4c     from official L
 4538496   45 47 4f 20 66 69 72 6d 77 61 72 65 0a 23 0a 23    EGO firmware # #
 4538512   20 54 68 65 20 55 2d 42 6f 6f 74 20 74 68 61 74     The U-Boot that
 4538528   20 73 68 69 70 73 20 77 69 74 68 20 74 68 65 20     ships with the
 4538544   6f 66 66 69 63 69 61 6c 20 66 69 72 6d 77 61 72    official firmwar
 4538560   65 20 66 6f 72 20 4c 45 47 4f 20 4d 49 4e 44 53    e for LEGO MINDS
<span style="background-color:#839496"><font color="#002B36">   /media/VMs/good-ev3dev.img                       7948206080  4538416   0%    </font></span>

</pre>

dlech, 2019-04-26:
I've narrowed down the version a bit. 1.5.26 seems OK and 1.5.28 is bad (there are no assets for testing 1.5.27).

This leads me to believe that 2c835437e9f2c9e688bb713af64c7cd083c469a8 introduced the problem.

thundron, 2019-04-26:
@dlech Thank you for the thorough analysis, we'll look into this

dlech, 2019-04-26:
Thanks, I hope that it can be prioritized since it is affecting many users (it is being reported on Facebook quite a bit).

lurch, 2019-04-26:
4538944 - 4538432 = 512 (i.e. the size of one disk-block), so maybe it's a simple off-by-one error? :man_shrugging:

dlech, 2019-04-26:
Good call:

https://github.com/balena-io-modules/etcher-sdk/blob/2811c7be2597706361778aca6932b70f41d141fd/lib/source-destination/zip.ts#L50

closed 2019-04-29 by zvin

laurensvalk, 2019-04-30:
Thanks for fixing this so quickly. Is there an estimated timeline for when this will make it into the next public release?

zvin, 2019-04-30:
@laurensvalk it's in 1.5.31, you can grab it from here https://github.com/balena-io/etcher/releases

laurensvalk, 2019-04-30:
That's awesome, thanks! Keep up the great work :+1: