balena-io/etcher #2755: grey screen with message "you don't have access to this resource"

mjrtom7, 2019-04-28:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

mjrtom7, 2019-04-28:
Etcher newbie here...
version 1.5.24 won't launch, giving me this error message and a gray screen.  Is there a way to rectify this issue?

lurch, 2019-04-29:
Similar to #2659 ?

mjrtom7, 2019-05-04:
Yes.  Gray screen, access message, and nothing else

markcusio, 2019-12-15:
As a non-root user I started Etcher from a linux terminal and got specific info of what resource I did not have access to.

john@johnwise /opt/balenaEtcher $ balena-etcher-electron
Error loading settings:
{ [Error: EACCES: permission denied, open '/home/john/.config/balena-etcher-electron/config.json']
  errno: -13,
  code: 'EACCES',
  syscall: 'open',
  path: '/home/john/.config/balena-etcher-electron/config.json' }
Gtk-Message: 22:58:43.950: GtkDialog mapped without a transient parent. This is discouraged.
ready-to-show: 21767.382ms

So I granted my user group "john" read/write to the folder and all sub folders/files:
/home/john/.config/balena-etcher-electron

lurch, 2019-12-16:
@markcusio Interesting! Why didn't your `john` user have read-write access to `/home/john/.config/` by default?

closed 2023-07-12 by dfunckt