balena-io/etcher #2756: GUI Windows turns blank grey

AMDphreak, 2019-04-29:
- **Etcher version:** 1.5.24
- **Operating system and architecture:** Win10 Pro 64-bit
- **Do you see any meaningful error information in the DevTools?** No (Did not know how to access this while the bug showed).

The window turns blank/grey after loading properly. This error happens with or without burning an image. It does happen while burning images. It is not 100% predictable.

[image]

thundron, 2019-04-29:
@AMDphreak You can check the dev tools' console by pressing `ctrl + shift + i`

MimbaMonkeyHouse, 2019-05-25:
I have exactly the same issue here. Can't even finish install process... Any solution?

thundron, 2019-05-26:
@MimbaMonkeyHouse could you paste here the logs you see in the devtools console mentioned above? `ctrl + shift + i` then focus the `Console` tab

jkuehl, 2019-05-28:
This maybe the same issue ?
[image]
[gui bug.txt](https://github.com/balena-io/etcher/files/3229290/gui.bug.txt)

thundron, 2019-05-28:
@jkuehl just released a new version for that, 1.5.43

jkuehl, 2019-05-28:
> 
> 
> @jkuehl just released a new version for that, 1.5.43
Looks fixed to me 👍 
[image]

closed 2019-05-29 by thundron

AMDphreak, 2019-05-31:
It was never a sure thing. It would randomly do this. I closed the app and didn't bother testing it for the random bug again. It's a high incidence rate, so I will add a log later.

Cabji, 2020-02-05:
I have a similar issue on my PC. After installing, I run Etcher (Windows 10) and all I get is a blank, dark grey/gray window titled Etcher.

From memory, I believe Etcher has _never_ worked correctly on this computer. It works on my wife's laptop, so I'm unsure if there's some underlying config problem on this computer that causes the trouble?

Etcher version installed: balenaEtcher-Setup-1.5.74

[image]

lurch, 2020-02-05:
Seems the 1.5.74 release is problematic - check #3057 (and its duplicates)