balena-io/etcher #2757: Etcher fails to write any file on any version AFAIK.

Inkertus, 2019-04-29:
Etcher version 1.5.30 (but also AFAIK ANY version)
Windows 10 version 1803 build 17134.706
Feren OS, Ubuntu, AFAIK ANY image...

I'm wondering if there's something weird about my computer setup/software/dependencies that breaks this software or something because I cannot get it to work and it doesn't give any meaningful errors.  Are there some DirectX packages I'm missing or...?  Failing that is there a debugging software I can run to help sort out the mystery?
[image]

lurch, 2019-04-29:
This seems to have happened a few times https://github.com/balena-io/etcher/search?q=writer+process+ended+unexpectedly&type=Issues

but I dunno if any solution was ever found.

Inkertus, 2019-04-29:
Doesn't appear that any fix was found but via one of the three identical posts (more or less) I went way back and tried a couple versions with results listed in the edit above.  I wonder what they changed between then and now.

lurch, 2019-04-29:
> I wonder what they changed between then and now.

LOTS! :wink:  https://github.com/balena-io/etcher/blob/master/CHANGELOG.md

thundron, 2019-04-29:
@Inkertus Probably trivial question: did you already try running it as admin?
Have you got any antivirus/malware or any file protection that might get in the way of etcher?

Inkertus, 2019-04-29:
I tried running it as normal (just double-click) and running as admin up front.  Same result across every version tested.  My anti-malware program is Malwarebytes and it gave no alerts.  Have not tried disabling it.

thunderhelper, 2019-05-03:
I'm getting the same error as in the screenshot above.  I tried the installed and portable version of etcher, running them as a regular user and as an admin.  

Running on: Windows 10 Pro Version 1803 Build 17134.706 64-bit

How do I get an old version of the app?  I can't seem to find the "old" downloads.
Thanks!

Inkertus, 2019-05-04:
I used this download (the portable one)> https://github.com/balena-io/etcher/releases/tag/v1.2.0

thundron, 2019-05-06:
@thunderhelper You can find older versions [here](https://github.com/balena-io/etcher/tags)
@Inkertus Did you try disabling Malwarebytes temporarily then re-run Etcher? If it works, you might consider adding an exception for it in the settings if it's possible

Inkertus, 2019-05-06:
Did that and no change.  Only thing that was different was that it failed _faster_ than before.  Otherwise the same result.

uchuugaka, 2019-05-23:
I started getting this non stop in .30 after updating Windows 10 today.  So I updated Etcher. Reinstalled too. No luck.it is broken.

Inkertus, 2019-05-23:
> I started getting this non stop in .30 after updating Windows 10 today. So I updated Etcher. Reinstalled too. No luck.it is broken.

You tried the above mentioned version (1.2.0)?  Only one that works for me.

uchuugaka, 2019-05-24:
I found it was a target USB flash drive that had Fedora 30 iso on it. 
Booting into Fedora I erased the drive and formatted it as NTFS and then booting back into Windows it worked fine. 
Probably some limitation of the app on Windows with foreign file systems.

lurch, 2019-05-24:
@uchuugaka Is the problem reproducible? i.e. if you flash a Fedora 30 iso to the USB drive again, does Etcher again become unable to flash the drive?

uchuugaka, 2019-05-24:
I’ll let you know.

uchuugaka, 2019-05-24:
So it’s not Fedora necessarily. 
Pop_OS! from System76 does it too. 
It seems to be that they’re both using encryption on a partition. 
Removing it in Gparted or Disks app in Gnome helps. 
I think I saw similar with SD cards for these on macOS too. 
Probably a missing thing in this app.

thundron, 2019-05-24:
@uchuugaka @Inkertus Could you please share the logs you see in Etcher's console? You can open it with `ctrl + shift + i`, then focusing the `Console` tab in the window that opens

splatkin, 2019-05-28:
right-click zip file > properties > check "unblock". Worked for me after that

ccroussore, 2019-06-09:
I receive exactly this same error when attempting to install Raspbian.

I have tried versions 1.5.45 and 1.5.34 of Etcher.  I have tried all three; I have also tried multiple USB adapters for Micro SD, and multiple Micro SD cards.  Finally, I have tried on two different computers (Windows 10 x64 and Windows 7 x64.

There is no ZIP file to unblock, so I am unsure what splatkin is talking about.  I did 'unblock' the .IMG files, not that it mattered.

Link to image files:
https://www.raspberrypi.org/downloads/raspbian/

Logs:
[-1560045544813.log](https://github.com/balena-io/etcher/files/3269098/-1560045544813.log)

splatkin, 2019-06-09:
After you download the zip file on Windows, right-click > properties > then
check "Unblock"
[image: image.png]

ccroussore, 2019-06-09:
splatkin :  That is not the solution to the problem I described above.  I have done it.  I did it again, just now, since you re-posted.  Same result.

It may be worth noting that the process does not fail until right at 99% of the burn.

[image]

lurch, 2019-06-09:
@thundron I see `Flash results Object` in the log that @ccroussore provided above, which seems to be coming from https://github.com/balena-io/etcher/blob/master/lib/gui/app/modules/image-writer.js#L250
Perhaps it should be using `JSON.stringify`to get more meaningful debug output?

fkromer, 2019-06-26:
Recently I had the same issue as well. 

Host OS: Ubuntu 18.04.2 LTS
Etcher: https://github.com/balena-io/etcher/releases/download/v1.5.50/balena-etcher-electron-1.5.50-linux-x64.zip
Image: https://github.com/rancher/os/releases/download/v1.5.2/rancheros-raspberry-pi64.zip
Target device: microSD card

thundron, 2019-08-14:
@fkromer Could you share your logs as well? (`ctrl + shift + i` to bring the console up)
@ccroussore Did you try updating etcher to the latest version?

Robert923, 2019-08-22:
Just had this issue on Windows 10 - it was the Windows 10 Ransomware Protection.
Windows Security | Virus & threat protection | Ransomware protection (at the bottom)
Then you can either:
- temporarily disable Controlled folder access (easier) or
- click on "Allow an app through Controlled folder access", and then choose "Add an allowed app". I find choosing the recently blocked apps is easiest. Then just select balena-etcher (NOT the installer, that won't help).

I hope that helps,

Robert

andyjsmith, 2019-08-22:
I am having the same issue, this has been happening to me for a while, I decided to try Etcher again after a few months and I still get this error. When I click flash, I click yes on the UAC prompt and then the writer process ends unexpectedly. I have tried this on multiple different images and on SD cards and USB drives.

Here is the log: https://pastebin.com/raw/rfzmFzAn

I am on version 1.5.56, I have "unblocked" the file as mentioned above, and I do not have the ransomware protection setting enabled as @Robert923 mentioned. Earlier versions of Etcher like 1.2 do work.

geraldpublic, 2019-09-01:
@Robert923 you are my hero - i have 2 pcs, on one of them i have the writer error issue, on the other on everything works fine, now i know why :-) the ransom protection was disabled on the working pc...
now it works on both of them again, thank you microsoft for changing settings with updates and nobody knows it :-)

closed 2019-09-25 by thundron

hardwarriore, 2020-06-09:
> right-click zip file > properties > check "unblock". Worked for me after that

thanks so much this one helped me

WorldWideWebDev, 2021-03-09:
Workaround on a MAC  sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher    enter your password and go ahead.. I wish someone could tell me the fix for this.. I know i caused it by clicking on a popup and did not allow the new installation to access my disks..  Grrrrr i dont know where to reverse what i have done and allow it once again...Help