balena-io/etcher #2761: Automated repository testing

fearphage, 2019-05-01:
It seems like there are several issues that have broken the (Debian?) repository:

[image]

Is there an automated (CI-able) way to confirm this is working so the owners don't have to find out from user bug reports?

lurch, 2019-05-01:
IIRC most of the "can't install from repo" issues tend to be due to temporary failures of an external GPG keyserver, which I don't think the Etcher team can do much about?

fearphage, 2019-05-01:
I scanned a few and they seemed to be due to missing files like the currently open #2458. That's why I thought it would be testable to make sure things are well-formed before we call a release "done".

lurch, 2019-05-01:
AFAICT the missing directory in #2458 is related to etcher-electron_1.4.4 and I think the Etcher packaging "stuff" has changed significantly since then?

thundron, 2019-05-14:
We changed quite a lot of stuff since then, and are still refining the process but generally if you see an error wrt downloading the packages from the repositories (deb and rpm), it's usually an issue with either ubuntu's apt(-get) or keyservers.
Missing versions are due to the fact that we don't release deb|rpm packages *as often*, but it's on my to-do list.
We also just released a new auto-update feature for the AppImage package, so that's generally our main focus

closed 2019-05-14 by thundron