balena-io/etcher #2770: 5 minute delay after selecting image on macOS Mojave

paulcrawford, 2019-05-07:
- **Etcher version:** 1.5.33 
- **Operating system and architecture:** macOS Mojave 10.14.4 
- **Image flashed:** ubuntu-18.04.2-4.9-minimal-odroid-n2-20190329.img 
- **Do you see any meaningful error information in the DevTools?** yes

The problem is that it takes almost five minutes to display the file name for the image to be flashed after clicking on the Select Image button. The following is the console output after the button was clicked. Originally I thought that it would not get the image at all since it was almost instantaneous in the old version of Etcher 1.4.4. I did wait for the image to be selected and then flashed it successfully so ultimately everything works. But to wait five minutes for the selection is unacceptable.

Mon May 06 2019 22:16:14 GMT-0400 (Eastern Daylight Time) Open image selector /Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 
({"applicationSessionUuid":"fc0f38d7-23d5-437d-9ad8-21180a341b1c","flashingWorkflowUuid":"e58f963f-58a9-4a0e-b1e7-1607785e18d3","sample":0.1})
Mon May 06 2019 22:21:05 GMT-0400 (Eastern Daylight Time) Select image /Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58
 ({"image":{"name":"ubuntu-18.04.2-4.9-minimal-odroid-n2-20190329.img","size":3607101440,"compressedSize":541137220,"isSizeEstimated":false,"hasMBR":true,"partitions":[{"offset":1048576,"size":134217728,"type":12,"index":1},{"offset":135266304,"size":3471835136,"type":131,"index":2}],"path":"/Users/paulcrawford/Downloads/ubuntu-18.04.2-4.9-minimal-odroid-n2-20190329.img.xz","extension":"xz","logo":false,"blockMap":false},"applicationSessionUuid":"fc0f38d7-23d5-437d-9ad8-21180a341b1c","flashingWorkflowUuid":"e58f963f-58a9-4a0e-b1e7-1607785e18d3","sample":0.1})

Can someone please determine where the problem is?

Thanks,

Paul Crawford

lurch, 2019-05-07:
I wonder if Etcher might be decompressing the `.xz` file to scan for some kind of metadata? :man_shrugging: 
Does Etcher work faster if you manually uncompress the .xz file first and then select the .img file in Etcher?

thundron, 2019-05-07:
@paulcrawford Is the UI unresponsive in general? Like taking 5+ seconds to go to the settings/back, opening the selection modal(s) and so on

zvin, 2019-05-07:
Is it slow only for this image?

paulcrawford, 2019-05-08:
Yes it does seem to be faster after manually uncompressing the .xz file first. This is the console output:

Tue May 07 2019 15:14:43 GMT-0400 (Eastern Daylight Time) Open image selector ({"applicationSessionUuid":"47727efd-d523-4745-9235-e430154f5346","flashingWorkflowUuid":"8b6d20d4-b93d-45ff-bd83-d86ec47e7c72","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 

Tue May 07 2019 15:15:09 GMT-0400 (Eastern Daylight Time) Select image ({"image":{"size":3607101440,"name":"ubuntu-18.04.2-4.9-minimal-odroid-n2-20190329.img","hasMBR":true,"partitions":[{"offset":1048576,"size":134217728,"type":12,"index":1},{"offset":135266304,"size":3471835136,"type":131,"index":2}],"path":"/Users/paulcrawford/ubuntu-18.04.2-4.9-minimal-odroid-n2-20190329.img","extension":"img","logo":false,"blockMap":false},"applicationSessionUuid":"47727efd-d523-4745-9235-e430154f5346","flashingWorkflowUuid":"8b6d20d4-b93d-45ff-bd83-d86ec47e7c72","sample":0.1})

Although it indicates a delta time of about 26 seconds rather than about 5 minutes the actual time to see the file name on the UI was a few seconds more, say about 30 seconds total from clicking “Open" on the file selection dialog to the file name being on the UI. In the older 1.4.4 version the file name took about 8 seconds to be visible on the UI.

So this is a significant improvement but surely still far too long. When the user is interacting with the Etcher to get it started, I would have though it should do nothing but retrieve the file name and show it to the user to verify the choice. It should definitely not start to uncompress the file since that clearly takes a long time possibly depending on the file size. My feeling was when I first saw this behaviour the Etcher was not going to work at all and since I did not wait five minutes it did not work. There is no indication (progress bar or anything else) to tell the user that something is happening. Why not do all of the decompression after the “Flash” button is clicked and only verify the image selection and drive selection to the user at the beginning. Then the user can do something else while waiting for the flashing to complete. This seems to be the way that the 1.4.4 version worked. Why was that changed in the 1.5.x versions?

Thanks,

Paul

lurch, 2019-05-08:
I wonder if it _might_ be related to #1248 ? That's simply a naive guess though. :man_shrugging:

paulcrawford, 2019-05-08:
@thundron I just checked opening preferences and it took about 13 seconds.

@zvin I just did a smaller image.xz for odroid c2 and it took about 3 minutes 45 seconds

zvin, 2019-05-09:
@paulcrawford I've tried opening this image in etcher 1.5.33 (md5sum `d2648abfa5052c8d3043936a53846ad9`) on an old mac mini.
The image name appears instantly.
Something else must be happening and eating all the cpu.

paulcrawford, 2019-05-09:
@zvin I did the same and you are right it is instantaneous. Then I restarted my MacBook Pro and it was instantaneous as well. I think you can close this issue.

Thanks and sorry for the bother.

zvin, 2019-05-09:
@paulcrawford I'd love to close this, as well as #2594 and #2740 .
I've never managed to reproduce this, all I have is 2 electron profiles and a video so I know it happens to some people.
It seems to be affecting osx only.

pior, 2019-12-29:
I can confirm the same behaviour (4 min with a compressed image, ~1min with uncompressed 3GB image).
But nothing else is loading my machine. 
There could be a bit of memory pressure though (if etcher tries to load the whole image).

etcher v1.5.70 
macOS 1.15.2

thundron, 2019-12-30:
@pior Did you try restarting the machine as previous users pointed out? I know it sounds cheesy, but as of now only OSX has this issue and we're clueless on how to reproduce.
It would be useful to track what you're doing up to the point where Etcher slows down that much, so we can maybe figure out what is eating resources

bingcheng1998, 2020-02-08:
I have the same issue. My Mac is 10.13.6:
[image]
very slowwwww for every step after one MacOS system upgrade, but I forget which one.

Not only etcher has this issue, but some other software, like nPlayer for mac, also very slow.

ealmonte32, 2020-04-21:
Add me to the list.. I also experienced this on version 1.5.71 so I updated to latest as of today thinking it would fix it, and same slowness persists, extremely slow to respond to clicks and opening anything, the flashing is not slow for me, whether decompressed image or not, it is the visual/graphics performance the issue..

thundron, 2020-04-24:
@ealmonte32 Might sound weird, but how long have you left your mac running without a system reboot? Did you try a quick reboot? Also @bingcheng1998 and @pior you might want to try it as well

bingcheng1998, 2020-04-24:
@thundron reboot doesn't help. But now I upgraded my OS to Mojave, and this problem disappears. So I think it only happens on High Sierra.

ealmonte32, 2020-04-24:
> @ealmonte32 Might sound weird, but how long have you left your mac running without a system reboot? Did you try a quick reboot? Also @bingcheng1998 and @pior you might want to try it as well

Yes I rebooted, cleared all the cache from both user and library paths, and looking at the libraries the process uses, nothing was producing high CPU at all, these are the system logs from my 10.13.x system:

_(I will just post as image instead because that way you can see the process and the message at once...)_

[image]
[image]
[image]
[image]

And then as you can see, the errors repeat as I repeat trying to use the app, these messages come up as I open the app, click on "Select Image", then click on the gear for Settings..

closed 2023-07-12 by dfunckt