balena-io/etcher #2772: firejailed Etcher does not work

rradar, 2019-05-08:
- **Etcher version:** 1.5.19
- **Operating system and architecture:** debian, 64
- **Image flashed:** -
- **Do you see any meaningful error information in the DevTools?** -

firejailing balena-etcher-electron is not successful out of the box. Just get a quite empty gui after 'ready-to-show: 4071.028ms'

[image]

I want to avoid giving network access to etcher before this one #2766 is fixed.

There is alternative software like opensnitch and douane to achieve this - but firejail is really light and it's just one command to put etcher into the sandbox

zvin, 2019-05-09:
Just tried with Etcher `v1.5.33`

```
Uncaught NodeError: A system error occurred: uv_interface_addresses returned Unknown system error 95 (Unknown system error 95)
    at Object.networkInterfaces (os.js:198:30)
    at getIPType (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/entities/Defaults.js:67:34)
    at new Defaults (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/entities/Defaults.js:45:21)
    at new Parser (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/entities/EventParser.js:8:14)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/dao/client.js:15:19)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/dao/client.js:259:3)
    at Module._compile (internal/modules/cjs/loader.js:693:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:704:10)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (internal/modules/cjs/loader.js:640:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/services/IPC.js:4:14)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/services/IPC.js:339:3)
    at Module._compile (internal/modules/cjs/loader.js:693:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:704:10)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (internal/modules/cjs/loader.js:640:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/node-ipc.js:3:13)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/node-ipc/node-ipc.js:23:3)
    at Module._compile (internal/modules/cjs/loader.js:693:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:704:10)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (internal/modules/cjs/loader.js:640:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:9328:18)
    at __webpack_require__ (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:20:30)
    at Object.<anonymous> (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:8941:13)
    at Object.<anonymous> (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:9322:30)
    at __webpack_require__ (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:20:30)
    at Object.<anonymous> (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:8649:21)
    at __webpack_require__ (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:20:30)
    at Object.<anonymous> (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:5937:40)
    at __webpack_require__ (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:20:30)
    at Object.<anonymous> (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:3899:1)
    at __webpack_require__ (file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:20:30)
    at file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:84:18
    at file:///opt/balenaEtcher/resources/app.asar/generated/gui.js:87:10
```

Requiring `node-ipc` fails, etcher won't work wothout that.

closed 2019-05-09 by zvin

thundron, 2019-05-09:
@rradar You could try to deny the analytics module calls only for now, or use a less-aggressive filter application that doesn't block *everything* but just calls to "the external world"? 🤷‍♂
Also be sure to use the latest `v1.5.33` as we included some fixes for analytics (which you can opt-out in the settings while we discuss switching them to opt-in)

rradar, 2020-03-18:
I could solve it finally!

Just did a `apt purge balena-etcher-electron`. Saved me over 400MB of disk space. My replacement which does no network calls at all is [usbimager](https://gitlab.com/bztsrc/usbimager)

rradar, 2020-03-19:
Censorship proudly presented by @balena