balena-io/etcher #2776: The writer process ended unexpectedly

KindlyPizza, 2019-05-10:
- 1.5.33 
- Windows 10 64 Bit
- recalbox.img.xz

receiving the writer process ended unexpectedly error with the following coming out of the console.

Experiencing this issue with multiple SD cards that were all working previously. Have also tried with a new SD card.

Successfully connected to IPC server: etcher-server-13472, socket root C:\Users\Bertrand Logan\AppData\Local\Temp\
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Image: C:\Users\Bertrand Logan\Downloads\recalbox.img.xz
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Devices: \\.\PhysicalDrive2
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Umount on success: true
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Validate on success: true
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Fri May 10 2019 15:11:33 GMT+0200 (South Africa Standard Time) Flash error ({"error":{"name":"Error","message":"Couldn't clean the drive, Command failed: diskpart s Bertrand Logan\\AppData\\Local\\Temp\\7b3dfa56610a.tmp\n (code 2147500037)","stack":"Error: Couldn't clean the drive, Command failed: diskpart s Bertrand Logan\\AppData\\Local\\Temp\\7b3dfa56610a.tmp\n (code 2147500037)\n    at Object.<anonymous> (C:\\Users\\Bertrand Logan\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\etcher-sdk\\build\\diskpart.js:92:23)\n    at Generator.throw (<anonymous>)\n    at rejected (C:\\Users\\Bertrand Logan\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\etcher-sdk\\build\\diskpart.js:20:65)"},"applicationSessionUuid":"ca287193-277d-48c0-9a37-d62dcff25cd1","flashingWorkflowUuid":"18821d90-f8f9-4fb3-b232-5be203210bbf","flashInstanceUuid":"235fb50d-a833-449f-b4c9-a9cacd462d80","image":"C:\\Users\\Bertrand Logan\\Downloads\\recalbox.img.xz","drives":[{"size":31914983424,"isVirtual":false,"enumerator":"PCISTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"F:\\, , , "}],"description":"SDHC Card","isUAS":false}],"driveCount":1,"uuid":"235fb50d-a833-449f-b4c9-a9cacd462d80","unmountOnSuccess":true,"validateWriteOnSuccess":true,"sample":0.1})
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Finish: 0
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Flash results Object
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Terminating IPC server
C:\Users\Bertrand Logan\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Fri May 10 2019 15:11:33 GMT+0200 (South Africa Standard Time) Open modal ({"name":"warning","applicationSessionUuid":"ca287193-277d-48c0-9a37-d62dcff25cd1","flashingWorkflowUuid":"18821d90-f8f9-4fb3-b232-5be203210bbf","sample":0.1})

lurch, 2019-05-10:
https://github.com/balena-io/etcher/blob/master/CHANGELOG.md says that v1.5.34 fixed "win32: fix running diskpart when the tmp file path contains spaces" so perhaps if you're lucky that'll have fixed your problem too? :crossed_fingers:

KindlyPizza, 2019-05-10:
That sorted it, Yay!

closed 2019-05-10 by KindlyPizza