balena-io/etcher #2777: Etcher : invalid extension

Pat054, 2019-05-11:
hello,
Debian installers on a raspberry pi3, I have this:

`pi@raspberrypi:~ $ sudo apt-get install balena-etcher-electron
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
N: « etcher.lis » dans le répertoire « /etc/apt/sources.list.d/ » a été ignoré car il utilise une extension non valable
E: Impossible de trouver le paquet balena-etcher-electron
pi@raspberrypi:~ $`

In the meantime, cordially

lurch, 2019-05-11:
See #2748

zvin, 2019-05-13:
The error is in the message: the file `etcher.lis` in `/etc/apt/sources.list.d/` has an incorrect `.lis` extension, it should be `.list`.

closed 2019-05-13 by zvin