balena-io/etcher #2779: debian repository key issue

entraide-net, 2019-05-12:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.ESyUbULGjn/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: failed to start the dirmngr '/usr/bin/dirmngr': Aucun fichier ou dossier de ce type
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.ESyUbULGjn/S.dirmngr' failed: Aucun fichier ou dossier de ce type
gpg: keyserver receive failed: Pas de dirmngr

lurch, 2019-05-12:
Looks like a problem with your apt-key setup, rather than a problem with Etcher? :man_shrugging: 

I did a quick search and found https://unix.stackexchange.com/questions/401547/gpg-keyserver-receive-failed-no-dirmngr

thundron, 2019-05-14:
Not sure I understand the error message, but it seems to be related to your setup/keyserver choice rather than Etcher. Try installing the `dirmngr` package as seen in the link suggested by @lurch and let us know, closing for now as it's unrelated to the application

closed 2019-05-14 by thundron