balena-io/etcher #2780: Flashing indicator jumps from 99%-100% to 47%, then after completion fails because of checksum mismatch

rytisbitcoin, 2019-05-12:
- **Etcher version:** 
1.5.35
- **Operating system and architecture:** 
Windows 7 Ultimate 64-bit
- **Image flashed:** 
http://wiki.fulmo.org/downloads/raspiblitz-v1.2-2019-05-02.img.gz
- **Do you see any meaningful error information in the DevTools?** 
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:02:19 GMT+0300 (FLE Daylight Time) Flashing 1 devices, 99% at 14.7 MB/s (total 14.7 MB/s) eta in 1.5669642857142858s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:02:20 GMT+0300 (FLE Daylight Time) Flashing 1 devices, 99% at 14.7 MB/s (total 14.7 MB/s) eta in 1.0669642857142858s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:02:20 GMT+0300 (FLE Daylight Time) Flashing 1 devices, 99% at 14.9 MB/s (total 14.9 MB/s) eta in 0.48855633802816906s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:02:21 GMT+0300 (FLE Daylight Time) Flashing 1 devices, 47% at 14.9 MB/s (total 14.9 MB/s) eta in 293.21026941140497s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:02:21 GMT+0300 (FLE Daylight Time) Flashing 1 devices, 47% at 14.9 MB/s (total 14.9 MB/s) eta in 292.58526941140497s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:02:22 GMT+0300 (FLE Daylight Time) Flashing 1 devices, 47% at 14.9 MB/s (total 14.9 MB/s) eta in 269.1171717643738s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:02:22 GMT+0300 (FLE Daylight Time) Flashing 1 devices, 47% at 14.9 MB/s (total 14.9 MB/s) eta in 258.59431355087844s with 0 failed devices

After flashing the image to 100%, it suddenly resets to 47% and goes to 100% again, then it completes successfully and goes to validation phase. After validation completes to 100%, it gives the message "Flash Complete! 1 Failed device". Mouseover shows "Source and destination checksums do not match". The Developer Tools show:

Mon May 13 2019 01:13:43 GMT+0300 (FLE Daylight Time) Verifying 1 devices, 99% at 20.3 MB/s (total 20.3 MB/s) eta in 0.5670103092783506s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:13:43 GMT+0300 (FLE Daylight Time) Verifying 1 devices, 100% at 20.6 MB/s (total 20.6 MB/s) eta in 0s with 0 failed devices
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:13:43 GMT+0300 (FLE Daylight Time) Validation error ({"applicationSessionUuid":"[redacted for privacy]","flashingWorkflowUuid":"[redacted for privacy]","flashInstanceUuid":"[redacted for privacy]","image":"C:\\Users\\Username\\Downloads\\raspiblitz-v1.2-2019-05-02.img.gz","drives":[{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"H:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"H:\\"}],"description":"Generic Storage Device USB Device","isUAS":false}],"driveCount":1,"uuid":"[redacted for privacy]","unmountOnSuccess":true,"validateWriteOnSuccess":true,"sample":0.1})
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Finish: [redacted for privacy]
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Flash results {results: {…}, cancelled: false}
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Terminating IPC server
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:13:46 GMT+0300 (FLE Daylight Time) Done ({"errors":[{"name":"Error","code":"EVALIDATION","device":"\\\\.\\PhysicalDrive2"}],"devices":{"failed":1,"successful":0},"status":"started","image":"C:\\Users\\Username\\Downloads\\raspiblitz-v1.2-2019-05-02.img.gz","drives":[{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"H:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"H:\\"}],"description":"Generic Storage Device USB Device","isUAS":false}],"driveCount":1,"uuid":"[redacted for privacy]","flashInstanceUuid":"[redacted for privacy]","unmountOnSuccess":true,"validateWriteOnSuccess":true,"applicationSessionUuid":"[redacted for privacy]","flashingWorkflowUuid":"[redacted for privacy]","sample":0.1})
C:\Users\Username\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Mon May 13 2019 01:13:46 GMT+0300 (FLE Daylight Time) Navigate ({"to":"success","from":"main","applicationSessionUuid":"[redacted for privacy]","sample":0.1})

lurch, 2019-05-13:
> After flashing the image to 100%, it suddenly resets to 47% and goes to 100% again

I think that's because there's no way to get the file-size of a gzipped image larger than 4GB. #638
What happens if you manually uncompress the image and write the `.img` file to your drive?

zvin, 2019-05-13:
Hello,

There is no way to know the uncompressed size of a gzipped file. The only thing we know is its uncompressed size modulo 4GiB.

So first Etcher shows you a progress bar thinking that the uncompressed size is 3.56GB.
When it reaches 3.56GB in the output, it realizes that the real uncompressed size is larger, it could be 7.85GB or 12.14GB or any value like `n * 4GiB + 3.56GB` where `n >= 1`.
So instead of guessing, it starts showing you the progress of the source read stream (we know where we are at reading the gzipped file) and shows you `47%` as it has read 47% of the original file.

I've checked writing this disk image to an SD card on both Windows and Linux with Etcher 1.5.35, I get no verification errors so I guess your SD card / usb stick is having issues, please try with another one.

closed 2019-05-13 by zvin