balena-io/etcher #2787: Privacyfriendly Etcher Fork?

rradar, 2019-05-15:
- **Etcher version:** recent
- **Operating system and architecture:** .nix
- **Images flashed:** many
- **Do you see any meaningful error information in the DevTools?** 

Is there any **maintained etcher fork** out there which **respects privacy** and **freedom of choice**?

This issue is somehow elated to #2766 and #2497 and many more closed ones..

closed 2019-05-24 by zvin

rradar, 2019-05-26:
:question: :grey_question:

rradar, 2019-06-08:
@zvin no interest?

thundron, 2019-06-08:
@rradar We're discussing this internally, we'll probably update the other issue though as there are more people that were interested in the privacy subject so to reach them as well

rradar, 2020-03-07:
No need to fork software which comes bundled with adware and spyware (namely balena etcher).

Alternativ in 2020 is just 250kb and does things right without leaking any data. [**usbimager**](https://gitlab.com/bztsrc/usbimager) respects your privacy and does it job.

Why should there be any need for bloatware (talking about more than 400MB till it finally runs/calls home) like etcher now? Maybe a good time to call it EOL @thundron?

rradar, 2020-03-18:
It's here! A **privacy friendly** etcher alternative!

But guess what: @balena-io doesn't like it and starts censorship :disappointed: 

> @balena-io balena-io just deleted a comment

Anyhow. The thing is called [**usbimager**](https://gitlab.com/bztsrc/usbimager) comes in a package of less than 1MB, does more things and does it better. No spyware, adware, bullshitware! :heart_eyes:

rradar, 2020-03-19:
Your **censorship has been detected and recorded** @balena-io

> https://web.archive.org/web/20200318180732/https://github.com/balena-io/etcher/issues/2787

gafarma89, 2020-05-23:
@rradar 

1- did you try unetbootin? it is also open source and it seems it is privacy friendly.

2- https://gitlab.com/bztsrc/usbimager --> is only developing by 1 contributer. I don't like this.

3- (Solution only for linux) you can start "etcher portable" with firejail. Firejail is very simple command line tool which blocks network for any app.

gafarma89, 2020-05-23:
> @rradar We're discussing this internally, we'll probably update the other issue though as there are more people that were interested in the privacy subject so to reach them as well

@thundron did you disccuss this internally? Is there any updates about privacy?

lurch, 2020-05-23:
> @thundron did you disccuss this internally? Is there any updates about privacy?

You've already left a comment on #2766 where @petrosagg (who works for Balena like @thundron does) has left multiple comments... :roll_eyes:

thundron, 2020-05-29:
@gafarma89 There's all the info you need in the other issues.
Locking