balena-io/etcher #2788: Bug: Featured Project while Flashing

Smilingouido, 2019-05-15:
- **Etcher version: 1.5.39** 
- **Operating system and architecture: Windows 10, x64** 
- **Image flashed: /** 

Hey,

just saw that the featured project disappears when I go while flashing into the settings and Exit them.

Smi

Picture:

[image]

thundron, 2019-05-17:
@Smilingouido Could you paste here the logs of the devtools console (`ctrl + shift + i`) ?

Smilingouido, 2019-05-17:
Where have I to press `ctrl + shift + i`?

thundron, 2019-05-17:
@Smilingouido While focusing Etcher window

Smilingouido, 2019-05-17:
Okay, thanks.

I will do it tomorrow when I have time.

Smilingouido, 2019-05-17:
But I have to flash again to get these logs, haven't I?

thundron, 2019-05-17:
@Smilingouido correct :) you have to reproduce the same situation as when you saw the issue

Smilingouido, 2019-05-29:
So, here are the complete results(sorry that I'm late.) : 

`C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Download the React DevTools for a better development experience: https://fb.me/react-devtools
You might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Warning: The "extend" API will be removed in the upcoming v4.0 release. Use styled(StyledComponent) instead. You can find more information here: https://github.com/styled-components/styled-components/issues/1546
console.(anonymous function) @ C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.39, Type = nsis
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed May 29 2019 14:55:58 GMT+0200 (Mitteleuropäische Sommerzeit) Application start ({"packageType":"nsis","version":"1.5.39","applicationSessionUuid":"49887eea-a449-40f8-b423-3a8fb04251c3","sample":0.1})
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed May 29 2019 14:56:05 GMT+0200 (Mitteleuropäische Sommerzeit) SafeWebview loaded ({"event":{"fromCache":false,"headers":{"Accept":"text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8","Upgrade-Insecure-Requests":"1","User-Agent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) balena-etcher/1.5.39 Chrome/66.0.3359.181 Electron/3.1.9 Safari/537.36"},"id":15,"method":"GET","referrer":"","resourceType":"mainFrame","responseHeaders":{"Accept-Ranges":["bytes"],"Cache-Control":["public, max-age=0"],"Connection":["keep-alive"],"Content-Encoding":["gzip"],"Content-Type":["text/html; charset=UTF-8"],"Date":["Wed, 29 May 2019 12:56:06 GMT"],"Etag":["W/\"3a720-16af90f90a0\""],"Last-Modified":["Mon, 27 May 2019 11:31:48 GMT"],"Server":["Cowboy"],"Transfer-Encoding":["chunked"],"Vary":["Accept-Encoding"],"Via":["1.1 vegur, 1.1 vegur"]},"statusCode":200,"statusLine":"HTTP/1.1 200 OK","timestamp":1559134565511.146,"url":"https://www.balena.io/etcher/success-banner/?etcher-version=1.5.39&api-version=2&optOutAnalytics=false"},"applicationSessionUuid":"49887eea-a449-40f8-b423-3a8fb04251c3","flashingWorkflowUuid":"cb803c9e-fa1e-4887-9e05-6e9e32a8ed06","sample":0.1})
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed May 29 2019 15:04:34 GMT+0200 (Mitteleuropäische Sommerzeit) Open image selector ({"applicationSessionUuid":"49887eea-a449-40f8-b423-3a8fb04251c3","flashingWorkflowUuid":"cb803c9e-fa1e-4887-9e05-6e9e32a8ed06","sample":0.1})
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed May 29 2019 15:04:40 GMT+0200 (Mitteleuropäische Sommerzeit) Select image ({"image":{"size":897581056,"name":"android-x86_64-8.1-r1.iso","hasMBR":true,"partitions":[{"offset":112640,"size":4194304,"type":239,"index":1}],"path":"C:\\Users\\samue\\Downloads\\android-x86_64-8.1-r1.iso","extension":"iso","logo":false,"blockMap":false},"applicationSessionUuid":"49887eea-a449-40f8-b423-3a8fb04251c3","flashingWorkflowUuid":"cb803c9e-fa1e-4887-9e05-6e9e32a8ed06","sample":0.1})
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed May 29 2019 15:04:42 GMT+0200 (Mitteleuropäische Sommerzeit) Flashing 0 devices, 0% at null MB/s (total null MB/s) eta in undefineds with 0 failed devices
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed May 29 2019 15:04:42 GMT+0200 (Mitteleuropäische Sommerzeit) Flash ({"image":"C:\\Users\\samue\\Downloads\\android-x86_64-8.1-r1.iso","drives":[{"size":30752000000,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"D:\\"}],"description":"SanDisk Ultra USB Device","isUAS":false}],"driveCount":1,"uuid":"58c89350-7cbb-425e-9be9-fc5b9a847e87","status":"started","flashInstanceUuid":"58c89350-7cbb-425e-9be9-fc5b9a847e87","unmountOnSuccess":true,"validateWriteOnSuccess":true,"applicationSessionUuid":"49887eea-a449-40f8-b423-3a8fb04251c3","flashingWorkflowUuid":"cb803c9e-fa1e-4887-9e05-6e9e32a8ed06","sample":0.1})
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Elevating command: C:\Users\samue\AppData\Local\Programs\balena-etcher\balenaEtcher.exe C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar
[... cut ...] b9a847e87","unmountOnSuccess":true,"validateWriteOnSuccess":true,"applicationSessionUuid":"49887eea-a449-40f8-b423-3a8fb04251c3","flashingWorkflowUuid":"cb803c9e-fa1e-4887-9e05-6e9e32a8ed06","sample":0.1})
C:\Users\samue\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed May 29 2019 15:06:54 GMT+0200 (Mitteleuropäische Sommerzeit) Navigate ({"to":"success","from":"main","applicationSessionUuid":"49887eea-a449-40f8-b423-3a8fb04251c3","sample":0.1})`

closed 2023-07-12 by dfunckt