balena-io/etcher #2790: Etcher 1.5.39 flagged for "Probably Bzip2Bomber" virus

diver-down, 2019-05-15:
- **Etcher version:** 1.5.39-x64
- **Operating system and architecture:** Windows 10 Enterprise
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** n/a

* * *

See attached screenshot -- the latest version of *both* balenaEtcher-Portable-1.5.39-x64.exe **and** balenaEtcher-Setup-1.5.39-x64.exe are being flagged for `Probably Bzip2Bomber`. The binaries were downloaded directly from https://www.balena.io/etcher/

Source: virustotal.com (for multiplatform scan to confirm)

I hope this report helps! :) 

* * *

[image]

lurch, 2019-05-15:
Hmmm, https://duckduckgo.com/?q=%2BBzip2Bomber only finds two results, both of which are virustotal.com result pages :confused:
Looks to me like one of the scanners is being over-sensitive?

thundron, 2019-05-17:
@diver-down Did this ever happen with previous versions?

diver-down, 2019-05-17:
> @diver-down Did this ever happen with previous versions?

Nope! This is the first version that I’ve had on my system (worth noting: I haven’t updated with every single release, so there are some versions awhile back that I never did download)

thundron, 2019-05-17:
@diver-down Well every new version of Etcher counts as a new application for various anti-malware and general file-protection softwares. I myself receive a warning everytime I open a new version on OSX.
Hopefully, auto-updates (which we added from version 1.5.30 on) will help with this type of warnings.
Meanwhile, I can only suggest ignoring those warnings for now if you download Etcher from a trusted source (releases are hosted on github, where the website points to as well)

Just curious: what version did you use previously, and could you try any of the versions from the one you had and the one you updated to?