balena-io/etcher #2797: Error: User did not grant permission

Ganondolf, 2019-05-23:
- **Etcher version:** 1.5.39-x64
- **Operating system and architecture:** Linux Mint 19.1 x64 with AppImageLauncher 1.2.2
- **Image flashed:** any
- **Do you see any meaningful error information in the DevTools?** yes, Error: User did not grant permission.

Yes, I have verified I'm typing my password correctly. Etcher goes back to the main window after I authenticate without doing anything

flashkiller999, 2019-06-03:
- Etcher version: 1.5.43-x64
- Operating system and architecture: Manjaro (All patches applied by 2019-06-03) with AppImageLauncher 1.2.2-3
- Image flashed: Any
- Do you see any meaningful error information in the DevTools? yes, Error: User did not grant permission.

Having the same problem.

thundron, 2019-08-14:
Try what this user suggested in his **Summary** section of this issue https://github.com/balena-io/etcher/issues/2881

closed 2025-03-18 by gbaso