balena-io/etcher #2807: [Feature Request] GUI selection for Legacy BIOS and UEFI

sevku, 2019-05-29:
The only tool that currently supports this is Rufus for Windows. Would be nice, if a Linux tool would support this as nicely as Rufus.

See Screenshot: https://rufus.ie/

And yes, I've seen the following issue, but that's not my request: https://github.com/balena-io/etcher/issues/2704

I've been (trying) to install Linux on a Surface Pro 4 which only supports UEFI and had some issues because of that. This would have greatly helped me.

lurch, 2019-05-29:
Sounds like that would involve modifying the image that gets written to disk, which AFAIK isn't something that Etcher supports yet? See also #718

sevku, 2019-05-29:
@lurch is there any way to make a UEFI bootable USB with Etcher? I sincerely did not manage to achieve that. The filesystem has been changed every time. Only Rufus has managed to do this.

lurch, 2019-05-29:
@sevku I have no experience with UEFI, and I'm not an Etcher-developer, only an Etcher-friend :slightly_smiling_face: 

My understanding is that Etcher copies the raw disk image file to your USB stick byte-for-byte, so if you want to use Etcher to make a "UEFI bootable USB" then you'll need an image file that is specifically set up to do that.

andreampiovesana, 2020-02-15:
very nice option! who can answer about it?

evprkr, 2021-03-16:
Is this being worked on at all?

thundron, 2021-03-22:
Etcher doesn't do any image manipulation by design: it just copies byte-to-byte the content of the source to the target. We have no plans on adding this functionality as of now

closed 2021-03-22 by thundron

Indus-01, 2023-01-12:
how to select bios or uefi mbr or gpt in etcher
like rufus on windows

or does linux have a rufus clone????