balena-io/etcher #2808: Error: No polkit authentication agent found.

jolupa, 2019-05-29:
- **Etcher version: 1.5.43** 
- **Operating system and architecture: Crux Linux 3.4** 
- **Image flashed: crux3-4, crux3-5 and Manjaro** 
- **Do you see any meaningful error information in the DevTools?
`Wed May 29 2019 16:03:39 GMT+0200 (CEST) Flash ({"image":"/home/jolupa/Baixades/crux-3.5-rc1.iso","drives":[{"size":7759462400,"isVirtual":false,"enumerator":"lsblk:json","logicalBlockSize":512,"raw":"/dev/sdb","error":null,"isReadOnly":false,"displayName":"/dev/sdb","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/sdb","busVersion":null,"isSystem":false,"busType":"USB","isCard":null,"isUSB":true,"devicePath":"/dev/disk/by-path/pci-0000:00:06.1-usb-0:2:1.0-scsi-0:0:0:0","mountpoints":[],"description":"CRUX-3.5-rc1 TOSHIBA TransMemory (CRUX-EFI)","isUAS":null}],"driveCount":1,"uuid":"004a9952-f2be-4b27-a643-a28b8988ee29","status":"started","flashInstanceUuid":"004a9952-f2be-4b27-a643-a28b8988ee29","unmountOnSuccess":true,"validateWriteOnSuccess":true,"applicationSessionUuid":"c5dcdfaf-cbfb-4d12-a14f-904d76d63bef","flashingWorkflowUuid":"e4fb4b20-2e50-4566-9b97-dd293190965e","sample":0.1})
/opt/balenaEtcher/re…js/dist/raven.js:58 Elevating command: /opt/balenaEtcher/balena-etcher-electron /opt/balenaEtcher/resources/app.asar
/opt/balenaEtcher/re…js/dist/raven.js:58 error Error: No polkit authentication agent found.
    at /opt/balenaEtcher/re…prompt/index.js:109
    at ChildProcess.exithandler (child_process.js:310)
    at ChildProcess.emit (events.js:182)
    at maybeClose (internal/child_process.js:961)
    at Socket.stream.socket.on (internal/child_process.js:380)
    at Socket.emit (events.js:182)
    at Pipe._handle.close [as _onclose] (net.js:596)

{"stack":"Error: No polkit authentication agent found\n    at Object.exports.createError (/opt/balenaEtcher/resources/app.asar/lib/shared/errors.js:253:17)\n    at Object.exports.createUserError (/opt/balenaEtcher/resources/app.asar/lib/shared/errors.js:292:18)\n    at sudoPrompt.execAsync.then.catch.catch.catch.catch (/opt/balenaEtcher/resources/app.asar/lib/shared/permissions.js:245:18)\n    at tryCatcher (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at /opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/catch_filter.js:34:37\n    at tryCatcher (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:512:31)\n    at Promise._settlePromise (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:569:18)\n    at Promise._settlePromise0 (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:614:10)\n    at Promise._settlePromises (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:690:18)\n    at _drainQueueStep (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/async.js:138:12)\n    at _drainQueue (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/async.js:131:9)\n    at Async._drainQueues (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/async.js:147:5)\n    at Async.drainQueues (/opt/balenaEtcher/resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)","message":"No polkit authentication agent found","description":"Please install a polkit authentication agent for your desktop environment of choice to continue","report":false,"image":"crux-3.5-rc1.iso"}`** 

I try different polkit agents like polkit-gnome xfce-gnome lxpolkit... All with the same problems, other installed programs are working perfectly.

lurch, 2019-05-29:
See also #2567

jolupa, 2019-05-29:
So reading the issue you point me I have to deduce that the problem is not solved? 'cause all the comments are reporting the same issue.
Let's wait for a fix! :)

thundron, 2019-05-30:
@jolupa Did you try running the appimage as root? Some users reported that's working for them, while we investigate on why it doesn't without root permissions

thundron, 2019-05-30:
Closing as duplicate of https://github.com/balena-io/etcher/issues/2567 btw, let's  continue there!

closed 2019-05-30 by thundron

jolupa, 2019-05-30:
> @jolupa Did you try running the appimage as root? Some users reported that's working for them, while we investigate on why it doesn't without root permissions

Yeah as root is working like a charm... I follow the issue you pointed! Thanks.