balena-io/etcher #2810: Please add .tgz file writing option

ampnanmathiyan, 2019-05-30:
- **Etcher version:**  1.5.39
- **Operating system and architecture:**  Mac OS 64 bit
- **Image flashed:**  Please add ".tgz" file writing option on Balena Etcher.
- **Do you see any meaningful error information in the DevTools?**  No

lurch, 2019-05-30:
I think this is an (indirect) duplicate of #2338 ?  (`.tgz` is simply a shortened version of `.tar.gz`)

Out of curiosity, which disk images are being distributed in a ".tgz" archive?

ampnanmathiyan, 2019-05-31:
Hi,

  The internal image we are making it ".tgz". So requested to add it.

lurch, 2019-05-31:
If it's an internal image you're making, IMHO it'd make much more sense to make it a `.img.gz` than a `.tgz`.

ampnanmathiyan, 2019-05-31:
Hi,

  .img.gz is working fine. If you make .tgz it will be really helpful.

Testato, 2020-09-02:
for examples the project PI-KVM is an Arch image for Raspberry and it is .tgz
Please add .tgz support

thundron, 2020-09-02:
You can select all files by changing filter in your OS file explorer. Closing

closed 2020-09-02 by thundron

Testato, 2020-09-02:
I tested on Ubuntu and Etcher do not show .tgz, instead the Ubuntu file explorer showing it.
So imho is the etcher interface that have limitation on some extension, at least on last Ubuntu version

lurch, 2020-09-02:
> for examples the project PI-KVM is an Arch image for Raspberry and it is .tgz

https://pikvm.org/download.html only shows `.img.bz2` files?