balena-io/etcher #2812: Updated etcher v1.5.43 is crashing on windows 10

sevenday4, 2019-05-31:
- v1.5.43 
-windows 10 64 bit 
- Image flashed: archlinux-2019.05.02-x86_64 iso
- **Do you see any meaningful error information in the DevTools?** 
﻿
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:02:55 GMT-0700 (Pacific Daylight Time) SafeWebview loaded ({"event":{"fromCache":false,"headers":{"Accept":"text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8","Upgrade-Insecure-Requests":"1","User-Agent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) balena-etcher/1.5.43 Chrome/66.0.3359.181 Electron/3.1.9 Safari/537.36"},"id":108,"method":"GET","referrer":"","resourceType":"mainFrame","responseHeaders":{"access-control-allow-origin":["*"],"age":["396"],"cache-control":["max-age=600"],"cf-ray":["4df4adf87cd7bb64-SEA"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Fri, 31 May 2019 00:02:32 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Wed, 29 May 2019 18:57:02 GMT"],"last-modified":["Mon, 20 May 2019 10:12:07 GMT"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["2"],"x-fastly-request-id":["ac97bc3ecdeb0588075bdb3d15400560a5132213"],"x-github-request-id":["AFB4:12D0:75578:7EB9D:5CEED8A0"],"x-served-by":["cache-sea1040-SEA"],"x-timer":["S1559260952.407135,VS0,VE0"]},"statusCode":200,"statusLine":"HTTP/1.1 200","timestamp":1559260975390.138,"url":"https://assets.balena.io/etcher-featured/index.html?etcher-version=1.5.43&api-version=2&optOutAnalytics=false"},"applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","flashingWorkflowUuid":"95c039eb-9014-41f1-9382-eb46da039f77","sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Successfully connected to IPC server: etcher-server-33472, socket root C:\Users\guypi\AppData\Local\Temp\
C:\Users\guypi\AppDa…js\dist\raven.js:58 Image: C:\Users\guypi\Documents\Arch Linux\archlinux-2019.05.02-x86_64.iso
C:\Users\guypi\AppDa…js\dist\raven.js:58 Devices: \\.\PhysicalDrive11
C:\Users\guypi\AppDa…js\dist\raven.js:58 Umount on success: true
C:\Users\guypi\AppDa…js\dist\raven.js:58 Validate on success: true
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:03:43 GMT-0700 (Pacific Daylight Time) Flash error ({"error":{"name":"Error","message":"Couldn't clean the drive, Command failed: diskpart s n (code 2147942405)","stack":"Error: Couldn't clean the drive, Command failed: diskpart s n (code 2147942405)\n    at Object.<anonymous> (C:\\Users\\guypi\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\etcher-sdk\\build\\diskpart.js:96:23)\n    at Generator.throw (<anonymous>)\n    at rejected (C:\\Users\\guypi\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\etcher-sdk\\build\\diskpart.js:20:65)"},"applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","flashingWorkflowUuid":"95c039eb-9014-41f1-9382-eb46da039f77","flashInstanceUuid":"963a2105-bd42-43dc-9aa1-e07641941903","image":"C:\\Users\\guypi\\Documents\\Arch Linux\\archlinux-2019.05.02-x86_64.iso","drives":[{"size":15669919744,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive11","error":null,"isReadOnly":false,"displayName":"K:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive11","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"K:\\"}],"description":"SanDisk Cruzer Glide USB Device","isUAS":false}],"driveCount":1,"uuid":"963a2105-bd42-43dc-9aa1-e07641941903","unmountOnSuccess":true,"validateWriteOnSuccess":true,"sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Finish: 0
C:\Users\guypi\AppDa…js\dist\raven.js:58 Flash results 
Object
C:\Users\guypi\AppDa…js\dist\raven.js:58 Terminating IPC server
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:03:44 GMT-0700 (Pacific Daylight Time) Open modal ({"name":"warning","applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","flashingWorkflowUuid":"95c039eb-9014-41f1-9382-eb46da039f77","sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:04:06 GMT-0700 (Pacific Daylight Time) Modal accepted ({"name":"warning","value":true,"applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","flashingWorkflowUuid":"95c039eb-9014-41f1-9382-eb46da039f77","sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:04:06 GMT-0700 (Pacific Daylight Time) Restart after failure ({"applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","flashingWorkflowUuid":"95c039eb-9014-41f1-9382-eb46da039f77","sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:09:45 GMT-0700 (Pacific Daylight Time) Open external link ({"url":"https://github.com/resin-io/etcher/blob/master/SUPPORT.md","applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:13:49 GMT-0700 (Pacific Daylight Time) Navigate ({"to":"settings","from":"main","applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:13:55 GMT-0700 (Pacific Daylight Time) Open external link ({"url":"https://www.balena.io/etcher?ref=etcher_footer","applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","sample":0.1})
C:\Users\guypi\AppDa…js\dist\raven.js:58 Thu May 30 2019 17:19:46 GMT-0700 (Pacific Daylight Time) Navigate ({"to":"main","from":"settings","applicationSessionUuid":"683ddbec-becb-4be9-8e42-51ebece3c807","sample":0.1})
﻿
The following is a snap shot of what the dialog box is saying:
[image]

I don't know if this issue shows up for the Linux versions yet. I will have to check on that tomorrow.​

zvin, 2019-06-03:
Hello @sevenday4 , thanks for opening an issue.

Did any previous version work?
The error here is `Command failed: diskpart s n (code 2147942405)`

Can you open a command line as Administrator, then launch `diskpart`
Inside `diskpart` you can try:
 * `list disk` to list the available devices;
 * `select disk <disk number from the list above>` to select the disk (please pay attention to select the correct disk);
 * `clean` to remove the partition table from the disk;
 * `rescan`

The above is basically what etcher does before flashing the drive on Windows (except it doesn't need to list the disks).

zeekless, 2020-09-09:
@zvin , thank you very much! I had the same issue, and your solution has worked!

closed 2020-09-09 by thundron