balena-io/etcher #2813: Download links broken

a13xch1, 2019-05-31:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

https://github.com/balena-io/etcher/releases/download/v1.5.24/balenaEtcher-Portable-1.5.24.exe

download links for etcher appear to be broken from github site, originally linked from https://www.raspberrypi.org/documentation/installation/installing-images/README.md

lurch, 2019-05-31:
Duplicate of #2811 ?

lurch, 2019-05-31:
Is there a problem with the way caching is configured on the Etcher website @thundron ?

pdcastro, 2019-05-31:
I was able to reproduce the issue with a Chrome extension called “[AdBlock](https://getadblock.com/)”. The issue was resolved by visiting the [Etcher website](https://www.balena.io/etcher/), clicking on the AdBlock icon and selecting _“Don’t run on pages on this site”._

The error and this workaround were also reported in the forums: https://forums.balena.io/t/broken-link-to-github/12752

closed 2019-06-01 by thundron

lurch, 2019-06-01:
Ahhh, so it seems people who are getting the "HTML version" of the site are getting old / broken links, and people who are getting the "Javascript-enabled version" of the site are getting the correct up-to-date links?
I guess a fix might be to modify the CI to update the "HTML version" of the site more regularly?

pdcastro, 2019-06-01:
@thundron, this issue has been closed, but isn't there room for any mitigation measure? If not to do something like @lurch suggested, then perhaps adding a discrete notice on the website in the lines of _"if download fails, check your ad blocker"._ But it's something for the Etcher team to decide, of course.

thundron, 2019-06-03:
@pdcastro I'll see what we can do, for now we have a PR open to bump the fallback version in case people have an adblock extension, will pick it up asap