balena-io/etcher #2814: Does Etcher support on ARM processor Raspberry Pi

llks, 2019-05-31:
- **Etcher version: 1.5.43** 
- **Operating system and architecture: Raspberry Pi Processor, ARM architecture** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

_May i know this software has any version that can run on ARM processor in Raspberry Pi? My project need to run on that platform.
_

lurch, 2019-05-31:
Duplicate of #2748 ?

thundron, 2019-06-03:
Yes it is @lurch 
@llks We are experimenting building on ARM but for now we only ship x86/x64 packages.
Closing for now

closed 2019-06-03 by thundron

llks, 2019-06-04:
Ok, thanks! Can't wait for the new release to the ARM platform as the market use so many ARM platform! May i know is there any target date for the ARM platform release?

thundron, 2019-06-04:
@llks As of now we're not working on the package itself so it's not really something we're planning to distribute (as of now). We'll surely let users know shall that happen!