balena-io/etcher #2816: Process failed and now the USB flash drive and now it's not recognized by any PC

AmitGolden, 2019-06-02:
- **Etcher version: 1.5.33** 
- **Operating system and architecture: Windows 10** 
- **Image flashed: Solus 4** 
- **Do you see any meaningful error information in the DevTools? No** 

I tried to download Solus 4 to a USB flash drive, the process failed, and now no computer recognizes it. 
I tried to edit the partitions with GParted on it to fix it, but it doesn't even show up on the drives list.
What can I do?

mametov, 2019-06-21:
If GParted doesn't see the flash drive, so what is the question to application? It seems like your flash has gone even if it lights/

joapps, 2019-10-10:
Try:  
Control Panel > Administrative Tools > Computer Management 
click on Disk Management under Storage 
insert the faulty USB device
it should appear as Removable Disk X (mine also displayed disk size and stated as "unallocated")

right click on the RIGHT-HAND SIDE FRAME of the box that appeared (mine showed 2 frames within the box -- right click on the frame on the right)
select New Simple Volume in the menu and go through the wizard

Note: wizard will assign a new drive letter (so it might be better to remove other removable devices before starting the wizard)

lurch, 2019-10-11:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

closed 2019-11-06 by zvin