balena-io/etcher #2820: v1.5.44 missing assets: broken link (NoSuchKey) for the 64-bit Linux installer

pdcastro, 2019-06-03:
Comparing the [v1.5.43](https://github.com/balena-io/etcher/releases/tag/v1.5.43) with the [v1.5.44](https://github.com/balena-io/etcher/releases/tag/v1.5.44) releases on GitHub, it looks like v1.5.44 is missing some files, including the `balena-etcher-electron-1.5.44-linux-x64.zip` file that linked from the website.

When users click on the website download link for the 64-bit Linux installer, they get a web page with:
```
<Error>
<Code>NoSuchKey</Code>
<Message>The specified key does not exist.</Message>
<Key>45055693/0e8d3400-8245-11e9-8657-527a67c9e921</Key>
<RequestId>24D49D96D3746DCD</RequestId>
<HostId>
r+bOCpfmTEWiCe0UWrhDHRyVPhvUKya5092GKdC85qfQS56BBX4b97PaXFROBinblq3qqOnIVxc=
</HostId>
</Error>
```

pdcastro, 2019-06-03:
For users attempting to install the v1.5.44 AppImage Linux installer, I can confirm that I was successful by downloading it directly from the GitHub releases page:
* https://github.com/balena-io/etcher/releases/download/v1.5.44/balenaEtcher-1.5.44-x64.AppImage
* https://github.com/balena-io/etcher/releases/tag/v1.5.44

(It's the zip file that is missing.)

luckman212, 2019-06-04:
[latest.mac.yml](https://github.com/balena-io/etcher/releases/download/v1.5.44/latest-mac.yml) points to an asset called `balenaEtcher-1.5.44-mac.zip` but that file does not exist...

lurch, 2019-06-04:
I guess this was fixed by #2823 ?

closed 2019-06-04 by thundron