balena-io/etcher #2821: Is a 32-bit AppImage meant to run on a 64-bit OS?

pdcastro, 2019-06-03:
Is a 32-bit installer (`balenaEtcher-1.5.43-ia32.AppImage`) supposed to run on a 64-bit OS? Not saying it should, just asking to confirm. A customer reported the following error that I was able to reproduce:

```
$ chmod +x *.AppImage
$ ./balenaEtcher-1.5.43-ia32.AppImage 
bash: ./balenaEtcher-1.5.43-ia32.AppImage: No such file or directory
$ uname -a
Linux osboxes 4.18.0-15-generic #16~18.04.1-Ubuntu SMP Thu Feb 7 14:06:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

The 64-bit image installed OK on the Ubuntu 64-bit OS. If the error is expected, this issue can be closed.

pdcastro, 2019-06-03:
Ideally (but most likely not a priority, and possibly a disproportional effort to the benefit), the error message would be something like _"32-bit installer not compatible with 64-bit OS",_ instead of _"No such file or directory"..._

lurch, 2019-06-04:
Just to clarify, the AppImages aren't an "installer", but a ready-to-run program; a bit like the Windows portable applications. I suspect changing the error message might need a change to the AppImage format itself? :man_shrugging:

thundron, 2019-06-04:
I'm not sure that it's possible, I've run into the same issue when attempting to do that some months ago. A quick explanation (along with some workarounds) can be found [here](https://github.com/RazZziel/PortableLinuxGames/wiki/Setup-a-64bit-system-to-run-32bit-appimages)
@lurch Is correct about the error: it's not something we can intercept unfortunately. Closing for now

closed 2019-06-04 by thundron