balena-io/etcher #2822: Windows asking to format drive many times, while validating image

frank26080115, 2019-06-04:
- **Etcher version:**  1.5.44
- **Operating system and architecture:** Windows 10, 64 bit, Intel
- **Image flashed:** jetbot_image_v0p3p0.zip for the Nividia Jetson Nano robot
- **Do you see any meaningful error information in the DevTools?** No, just the progress of the validation

Basically, during the image validation phase, Windows asked me if I wanted to format my drive 11 times. Each time, it came up with a different letter.

I am using a USB 3.0 microSD card reader/writer to do this. The card is a 64 GB microSD card.

I am hoping something in Etcher can suppress these dialogs, or prevent Windows from attempting to mount the drive during validation.

lurch, 2019-06-04:
Duplicate of #2210 ?

frank26080115, 2019-06-05:
Probably yes, sorry, handle it however you guys handle it

I encountered a complete validation failure when the drive seemingly disconnected, it seems like that was in #2210 as well

lurch, 2020-06-12:
Related to #3197 ?

closed 2020-08-19 by thundron