balena-io/etcher #2825: Last available version in repository is 1.5.19 (from march 2019)

rradar, 2019-06-08:
- **Etcher version:** 

Lattest version from offical debian repo: **1.5.19**
https://dl.bintray.com/etcher/debian/pool/e/etcher/

- **Operating system and architecture:** 

debian based

- **Image flashed:** 

you don't need to care

- **Do you see any meaningful error information in the DevTools?**

rradar, 2019-06-08:
....and it's june now

peliopoulos, 2019-07-03:
July!

thundron, 2019-07-22:
Updated to 1.5.51 💪

closed 2019-07-22 by thundron

rradar, 2020-03-18:
Not needed anymore! Save disk space and purge etcher :laughing: 

[**usbimager**](https://gitlab.com/bztsrc/usbimager) comes in a package of less than 1MB (like 400 hundred(!) times less than etcher), does more things and does it better. No spyware, adware, bullshitware! :heart:

lurch, 2020-03-18:
Does usbimager allow you to flash RPi Compute Modules? Does usbimager allow you to flash to multiple drives at once?

rradar, 2020-06-25:
> Does usbimager allow you to flash RPi Compute Modules? Does usbimager allow you to flash to multiple drives at once?

No idea because I never used such features in my live.

What matters to me is **privacy, size** and **speed** - usbimager outperforms etcher in all these :rocket: