balena-io/etcher #2826: Checksums not to be found?

AreDubya, 2019-06-08:
- **1.5.45** 
- **Windows 10** 
- **No images flashed** 

Hello, there are a few references to verifying these releases in different issues, but the current release's checksum is either an HTTP 406 error or Not Found. 

https://github.com/balena-io/etcher/releases/download/v1.5.45/SHASUMS256.txt

Are the checksums still on GitHub? If so, where?

lurch, 2019-06-08:
I had a quick look at https://github.com/balena-io/etcher/releases/tag/v1.5.45 and it seems like there are checksums inside the `latest*.yml` files?

@zvin Maybe it'd be useful to have the CI process collate them into a single `SHASUMS512.txt` ?

AreDubya, 2019-06-09:
A central location would be ideal, but that's not the checksum in latest.yml, it's the checksum in binary piped through base64.

This is the sha512 checksum : 
`86e6f73a03a96d073579bbb20b7b25a346ab6b20bcea204bb4b26b36370f90e35217f342de43e4b92fd25b73dbb3ff1165ce27b2800967211ddb0bc5fb98daa7`

The value in latest.yml is the checksum (in binary) encoded in base64 : 
`openssl dgst -sha512 -binary balenaEtcher-Setup-1.5.45.exe | openssl enc -base64`

| Source  | Output |
| ------------- | ------------- |
| latest.yml | hub3OgOpbQc1ebuyC3slo0arayC86iBLtLJrNjcPkONSF/NC3kPkuS/SW3Pbs/8RZc4nsoAJZyEd2wvF+5japw== |
| `sha512sum`  | 86e6f73a03a96d073579bbb20b7b25a346ab6b20bcea204bb4b26b36370f90e35217f342de43e4b92fd25b73dbb3ff1165ce27b2800967211ddb0bc5fb98daa7 |
| `openssl dgst -sha512 -binary balenaEtcher-Setup-1.5.45.exe \| openssl enc -base64` | hub3OgOpbQc1ebuyC3slo0arayC86iBLtLJrNjcPkONSF/NC3kPkuS/SW3Pbs/8RZc4nsoAJZyEd2wvF+5japw== |

IMO, ideally the `README.md` should have the instructions spelled out for binary and piping to base64 with a link to the `SHASUMS512.txt` displayed prominently, to make the info accessible for all levels of enthusiasm.

FlexMcMurphy, 2021-11-20:
Can there just be a page where you can view the checksums?
Like from the [Putty download page](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) there are links at the bottom of the page to Checksum files.

gyrusdentatus, 2021-11-21:
what's the big deal? Such a trivial thing to post checksums along with the releases ... on macos I only can see sha512 for a zip file not the actual .dmg file 

fix this please, it's trivial

BamBuBang, 2021-11-21:
How do I verify balenaEtcher hash.
I need to burn a tails iso from a checksum verified application like balenaEtcher. 
This is what I have so far:
balena-etcher-electron-1.7.0-linux-x64.zip
D3EEE4078C6AC972D5434666DB4E6A66B3F67F32E6C8E2D0521FD5CE2A05835A3A30DEA878BB19F7D406D065138569A5C8BF104A3ED79002DA6056FB31EF0ECF
I get the above hash when I entered the file into Quickhash v3.3.0.
How can I verify the hash?
Is there another recommended  burner I can use with Linux?
Any advice is welcome.

sid-the-sloth, 2023-07-13:
Please provide SHA256 checksum files, it can be an automated step during build.
See how VSCodium or Mdview are doing it.
Thanks!

gyrusdentatus, 2023-07-21:
VSCodium what is it? Is it like VIM? @balena-ci @balena-deploy hire me, I will do this dirty work for minimum wage coz I dig your products and woud love to work more on them ! :)) one call and you’ll see? 

> Please provide SHA256 checksum files, it can be an automated step during build.
> See how VSCodium or Mdview are doing it.
> Thanks!

sadtank, 2023-07-27:
Someone over there thinks it's a good idea to obfuscate hashes with base64... or just use base64 as a "hash"... I don't see any clear instructions on how the .yml values are generated, so I can't even reproduce whatever that value is (let alone verify that process actually purports to do what they intend). So whatever the .yml values are there for, it's a waste of time. I can't trust the integrity of your binaries and I seriously doubt the security competence of whoever thought this was a good idea. It takes all of what, 10 minutes to generate and post hashs of all your stuff? If you don't value the trust of your users, I wish you a speedy failure and may a more trustworthy utility absorb your user base.

dfunckt, 2023-07-27:
@sadtank thanks for your input. You're entitled to your opinion but this is not a forum or IRC -- please be civil and refrain from posting unconstructive comments.

sadtank, 2023-07-27:
Been a documented issue for 3 ish years? What's being done?

dfunckt, 2023-07-27:
Not sure what the expectation is here -- that someone at balena should jump to cater to every issue reported? Evidently, it is currently not a priority, thus it isn't being worked on. Honestly, we've got much bigger fish to fry. If it's so annoying to you, please take a moment or two and share a PR.

Again, this is not a forum for lightweight discussion -- if this continues I'll lock the issue since clearly there's not much further to discuss than "this feature does not exist, please add it".

sadtank, 2023-07-27:
It's not really a feature though... It's really a 5 minute operation as part of publishing a new version. So instead of tracking here as a dev issue maybe it's a simple process improvement... Won't be hard to add that to the publishing process, right? No dev needed.

It's a long shot: As a member, would you be able to run a single command for me against a known-good binary on your end... that would be something. Can you attest that one/both of these hashes are valid for this image? (or post those of any other recent x64 linux version?

`$ sha256sum balenaEtcher-1.18.11-x64.AppImage 
f87bbbd1439c98b5f874f39810103669d05d5ff0bc472a7fd7b6ef8be8d47d50  balenaEtcher-1.18.11-x64.AppImage

$ sha512sum balenaEtcher-1.18.11-x64.AppImage 
1c1b2bbf526a38f29651df1ab95520b40728a45f04edbfca074a757524a206c53f71788abc59bc73d4d7a13690c5ecc37744e1a5bac94fee31e581f953bfc21e  balenaEtcher-1.18.11-x64.AppImage`

sid-the-sloth, 2023-07-27:
I would keep the sha256 or sha512 hashes in a normal hash file (NOT base64 encoded) in plaintext.

Format being, for example:

```
75572bd2a0c6f79402f5acb549b86cf48c421b3a4bbde8f65d19a37b6f5f5a14  appimagename-1.0.94-linux.AppImage
202cb7ab9cef7cc8ed62fd71778da3e526c4fc82895ebdbbbab590e2f99828f3  appimagename1.0.94-windows.Appimage
```

gyrusdentatus, 2023-07-27:
> Not sure what the expectation is here -- that someone at balena should jump to cater to every issue reported? Evidently, it is currently not a priority, thus it isn't being worked on. Honestly, [...]

open a PR to fix your CI ? Might work if you’re using github actions I guess (not sure and not on pc to check right now). 
Seriously, this should be a standard especially for the stuff that you guys are pushing. I will gladly fix this thing for you if you provide tiny amount of details for me reg your pipelines for releases.

sadtank, 2023-07-27:
I donnow... the timeline of this issue, these answers... almost like they don't _want_ our trust... lol. oops, sorry, not an IRC...

gyrusdentatus, 2023-07-27:
lm tell you a secret - this is actually a netcat chat over websockets !

> I donnow... the timeline of this issue, these answers... almost like they don't _want_ our trust... lol. oops, sorry, not an IRC...

dfunckt, 2023-07-27:
Thanks for your input folks -- point taken. We'll take a look as soon as we get a chance. I'm locking this convo.

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)
closed 2023-12-21 by dfunckt