balena-io/etcher #2833: Etcher broken on macOS Catalina

nathfreder, 2019-06-24:
- **Etcher version:** 1.5.50
- **Operating system and architecture:** macOS Catalina 10.15 Beta (19A487l)
- **Image flashed:** http://director.downloads.raspberrypi.org/raspbian_full/images/raspbian_full-2019-06-24/2019-06-20-raspbian-buster-full.zip
- **Do you see any meaningful error information in the DevTools?** 
```
{"stack":"Error: EPERM: operation not permitted, open '/Users/nathfreder/Downloads/2019-06-20-raspbian-buster-full.img'","message":"EPERM: operation not permitted, open '/Users/nathfreder/Downloads/2019-06-20-raspbian-buster-full.img'","name":"Error","code":"EPERM","syscall":"open","errno":-1,"image":"2019-06-20-raspbian-buster-full.img"}
```

nathfreder, 2019-06-24:
I also get a message on-screen: 
[image]

coreyasmith35, 2019-06-24:
Having this same issue.

nathfreder, 2019-06-24:
**Workaround:**
Moving your image to the balenaEtcher bundle seems to do the trick.

willswire, 2019-06-25:
Workaround listed above works for me as well.  I think this error is in part to the recent changes made by apple (see [here](https://www.apple.com/macos/catalina-preview/)) regarding disk access permissions. "... apps must now get your permission before directly accessing files in your Documents and Desktop folders, iCloud Drive, and external volumes, so you’re always in control of your data."

mydudechris, 2019-06-26:
@nathfreder suggestion worked for me as well.

SimplyGardner, 2019-06-28:
@nathfreder  Suggestion worked for me too! Times like these, I realise there really is no alternative to Etcher!

DG12, 2019-06-30:
Is this related to "Long directory/filename seems to fail" #2843

xcvista, 2019-07-01:
Maybe Balena should codesign the app with disk access? Apple as been tightening the sandbox.

nathfreder, 2019-07-01:
@xcvista That would be a good idea. It's required for Catalina.

cameroncooke, 2019-07-13:
Above approaches didn't work for me, what did:

```bash
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
```

Hope this helps.

nathfreder, 2019-07-13:
@cameroncooke That's way simpler than my approach. 😀

livefeed, 2019-08-01:
@cameroncooke Thank you, this worked for me.

MiroHibler, 2019-08-10:
macOS Catalina 10.15 Beta (19A526h) doesn't even allow the app (>= v1.5.51 as far as I know) to start due to "check against the malicious software".

v1.5.33 can start, but `sudo` hack doesn't work anymore.

nathfreder, 2019-08-10:
@micopiira so the app doesn't start at all, or doesn't start with the `sudo` hack?

thehappydinoa, 2019-08-11:
`v1.5.52`  is currently working on `10.15 Beta (19A512f)`

thundron, 2019-08-12:
For macOS Catalina 10.15 I believe we'll have to notarise the Etcher application package as it'll be a requirement for all applications from 10.15 on, which we're already preparing for

Betabao, 2019-08-21:
> Above approaches didn't work for me, what did:
> 
> ```shell
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
> 
> Hope this helps.

this works for me.

kaushaltrivedi, 2019-08-30:
> Above approaches didn't work for me, what did:
> 
> ```shell
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
> 
> Hope this helps.

this worked for me!!!

ucdailoi, 2019-09-02:
Wont work for my mac..... What started out as an attempt to load Home Assistant onto a SD card has become a nightmare of chasing shadows. I have to give up after 3 hours of stuffing around... What now is the question

iglesias294, 2019-09-11:
Moving images inside the etcher package not working on Catalina Beta 8. Launching with SUDO does work. Log below: 

Sandbox: balenaEtcher Hel(9055) System Policy: deny(1) file-read-data /dev/rdisk2
**Violation:       System Policy: deny(1) file-read-data /dev/rdisk2** 
Process:         balenaEtcher Hel [9055]
Path:            /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper.app/Contents/MacOS/balenaEtcher Helper
Load Address:    0x1098a9000
Identifier:      io.balena.etcher.helper
Version:         1.5.56 (???)
Code Type:       x86_64 (Native)
Parent Process:  bash [9054]
Responsible:     /System/Library/Frameworks/Security.framework/authtrampoline

kschembri, 2019-09-17:
@cameroncooke thank you, launching with SUDO works for me

kapong, 2019-09-18:
@cameroncooke  @kschembri thank you, `sudo` works for me too

netyrk, 2019-09-20:
> Above approaches didn't work for me, what did:
> 
> ```shell
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
> 
> Hope this helps.

10.15 Beta (19A558d)

sudo solution works for me.  Other potential solutions provided did not work.

zaltal, 2019-09-20:
Workaround fixed it for me swell on OS X Catalina. Thanks nathfreder!

Hansaplast, 2019-09-25:
I bumped into this "issue" while looking for a solution for my own application ([ApplePi-Baker](https://www.tweaking4all.com/software/macosx-software/applepi-baker-v2/)).
Since we're all here with the same goal in mind (writing SD cards for RaspberryPi), I thought I'd help out and post my fix.

Under Catalina one must add the application to the security exceptions:
Go to “**System Preferences**” -> “**Security & Privacy**” -> “**Privacy**“.
Add the application to the list of “**Full Disk Access**“ items, and you should be good to go.
No need to run as sudo.

cameroncooke, 2019-09-25:
I had tried this in the betas and it still didn't work.

Also shouldn't the system prompt you to allow this as soon as it [Etcher]
attempts to make full-disk access?

Hansaplast, 2019-09-25:
Sorry to hear that, I just tested it again with Catalina beta 9 (19A573a) and it worked on my Mac. 
I noticed that I had to reboot  Mac (since I use a Privileged Helpertool, which needs to be restarted as well and keeps being "available" in the background, so the changes may not have affected it yet - I suspect).

I'm sure the Etcher devs would agree with me that it would be appropriate if the system would actually prompt the user (either asking for permission, or pointing out why it didn't work). But it doesn't seem to do it for "Full Disk Access". Still looking for a solution though ...

I've been looking for info all day long already and have not been able to find a proper answer.

ReubenBTalbott, 2019-09-28:
As a temporary solution for anyone wanting to run etcher on Catalina easily, I have made an Apple script to run etcher with administrator privileges. You can just run the script or turn it into an app which is what I have done personally. 

`display dialog "Start balinaEtcher with Admin Privilages?" buttons {"Start", "Cancel"}
do shell script "Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher" with administrator privileges`

idoodler, 2019-09-29:
@thundron You are right. macOS Catalina requires applications to be notarized, we are facing a similar issue. Have you had success with notarization yet?

Hansaplast, 2019-09-29:
I managed to notarize ApplePi-Baker successfully, but it will not resolve the disk access issue, as it only notarizes the distribution DMG or ZIP file ([reference](https://developer.apple.com/documentation/security/notarizing_your_app_before_distribution)).

You will need to get an application specific password at https://appleid.apple.com/.
This app specific password can be reused for following versions of the application.

Then with that code you have to notarize with:

`xcrun altool --notarize-app --primary-bundle-id "com.compay.appname" -u "yourappleid@email.com" -p "app-specific-password" -t osx -f "yourappdmg.dmg"`

This will produce a Request UUID.

Once notarized, you will get an email confirming that it has been notarized.
You can also check your status from Terminal:

`xcrun altool --notarization-history 0 -u "yourappleid@email.com" -p "app-specific-password"`

After being notarized you will need to "staple" your DMG (or ZIP) before you can distribute it.

`xcrun stapler staple “yourappdmg.dmg"`

This is how it worked out for me, and does not seem to be sensitive to using non X-Code development tools. 
Hope this is helpful.

ReubenBTalbott, 2019-10-08:
MacOS Catalina was released today, is there any update on getting balinaEtcher running properly?

p-i-, 2019-10-08:
Just trying this out on Catalina.

It works with the `sudo` trick (above).

My guess is that devs need to add a couple of permissions to the `info.plist`

EricGoodrich, 2019-10-08:
I was having trouble flashing an image while running the **Official Release of macOS Catalina** (10.15.0). 

I tried the:
"Go to “System Preferences” -> “Security & Privacy” -> “Privacy“.
Add the application to the list of “Full Disk Access“ items" from above and that didn't work either.

Then I did the "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher" and I was able to flash my image just fine. 

Perhaps balenaEtcher app needs to be updated. In the meantime, the sudo workaround seems to work fine.

Thank you everyone for the discussion and I will be following this thread for future developments. 

~e
🍻

boltomli, 2019-10-09:
It also has the access to removable media. When running inside a terminal, the access request window can pop up correctly then it'll work well.

tymoteuszrogalewski, 2019-10-09:
Yesterday update MacBook to Catalina. BalenaEther 1.5.56 stops work ;)
Terminal gives popup with access to Downloads folder and ok, but balena not. Just writing error message. Adding full access to disk under Privacy preferences gives nothing.
This workaround with
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
...works great for now. But must run with this method every time. Access privileges in remembered only via terminal run and not work with later common app running by mouse clicking.
Hope soon will be balena update with fix.

fortyKnoxy, 2019-10-09:
> Above approaches didn't work for me, what did:
> 
> ```shell
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
> 
> Hope this helps.

This method worked for me, thanks a lot.

tgwaste, 2019-10-10:
Worked for me as well.  Thanks!

magiconair, 2019-10-10:
The workaround is two-fold: (tested with 1.5.58 on macOS 10.15)

1. Go to `Security & Privacy` and allow etcher to run. There should be a warning that etcher was blocked but you can allow to run it anyway. This will get rid of the initial warning that the app has to be updated. However, then flashing still fails with the error message that the image is corrupt.
2. Run etcher with `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher` allows you to flash as before.

alexmuntean, 2019-10-12:
i managed to flash by copying inside balenaetcher app itself and selecting it from there and disabling system integrity protection (command+r while restarting -> terminal -> csrutil disable)

magiconair, 2019-10-12:
The system integrity protection is there for a reason. I don’t think it is a good idea to disable it and to explain this as a way to work around the problem if there are other alternatives. 

So try all the other approaches first and only if they all fail try the system integrity approach.

stacywebb, 2019-10-13:
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher

works on version 10.15.1 Beta (19B68f)

Black0wI, 2019-10-13:
Sudo workaround works on macOS 10.15 (19A583) with balenaEtcher 1.5.57.
Thank's

Hammid, 2019-10-13:
Yes. God bless the `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher` workaround!

mydudechris, 2019-10-13:
Still having this issue. Running 1.5.58 on OS X 10.15 Beta (19A582a)

fraban, 2019-10-14:
Hi all - I build a 2-line shell command and put this into "Automator" ... a simply Wrapper - now I can start it by mousclick. 
its easy: 

1. Build a shell-script with using your password for "sudo - command" and  Balena Etcher standard install.

Open your Textedit and type   ( replace <YOURPASSWORD> with your password
```
#!/bin/zsh
echo <YOURPASSWORD> | sudo -S /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
```
Save it as   "WrapperBalena.sh" in your $HOME directory Applications folder  ($HOME/Applications)

2. make the scrip executable - Open "Terminal" and type 
cd $HOME/Applications
chmod a+x WrapperBalena.sh

3. Now open Automator and look for "Shell-Script Execution" 
and insert - s.Screenshot
[image]

4. Save it on your desktop as e.x. Belena-starter   and your done ... 
Doubleclick the new automator-file and BelenaEtcher starts as "root" and works on Catalina...  

It´s only a fix - not a final solution ! 
-frank

gh0st, 2019-10-16:
Sudo trick now does not work with macOS 10.15 (19A602).

GiuseppeLaurenza, 2019-10-17:
> Sudo trick now does not work with macOS 10.15 (19A602).

Same problem, with sudo flash starts, but at the end it says that checksum of image and usb is different

muddydev, 2019-10-20:
> Above approaches didn't work for me, what did:
> 
> ```shell
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
> 
> Hope this helps.

Still working

jorgemanzo, 2019-10-20:
Sudo trick is working on Catalina 10.15 and Etcher 1.5.59

chrschmitz, 2019-10-21:
Go to the Application folder and open the program with a right-click: Open

daballiemo, 2019-10-21:
> Go to the Application folder and open the program with a right-click: Open

Same result 
[image]

gh0st, 2019-10-21:
**FYI, just saying that the app does not work does not help. Please include your version of macOS with** `sw_vers` in a terminal window or the hard way by clicking the Apple icon in the top left hand corner of finder and clicking _About This Mac_ and then the _System Report..._ button in the _Overview_ tab. Then clicking _Software_ in the navtree on the left hand side of the _System Information_ window.

daballiemo, 2019-10-21:
fyi I already put all information (including MacOS version in my first post. But to be clear again:

MBP2018-2:~$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.15
BuildVersion:	19A602
MBP2018-2:~ $

balenaEtcher version: 1.5.59

Etcher won't start either from Launcher (see screenshot in my previous post) of via sudo in terminal session with the following command: 
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
The result I get is: Killed

can't make it anymore clear than that.

ideologysec, 2019-10-21:
@daballiemo  try `xattr -d com.apple.quarantine /Applications/balenaEtcher.app` in Terminal, and then opening the app either via sudo in Terminal or double-clicking in the Finder.

RainmakerRaw, 2019-10-23:
Also getting 'Something went wrong... check the archive is not corrupted' for every single ISO I try - over a dozen so far (Ubuntu 19.10, Void Linux, Pop_OS! 19.10, NomadBSD etc). I tried a complete removal (AppCleaner.app) and reinstall of the latest Version 1.5.59 (1.5.59), but still the issue persists. I'm on macOS Catalina 10.15 (with supplemental update). 
[image]

tejavegesna, 2019-10-23:
After updated to Catalina. I have reinstalled i open up. It says macos says its a mailware. So i go to System Preferences >> Security & Privacy >> Open Anyway for Etcher Balena
And Try to reopen again. Close the opened Window.

Open a New terminal window and type `xattr -d com.apple.quarantine /Applications/balenaEtcher.app`
And then `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`
Type password for user.
Then Select image & USB drive Then press **Flash**
Then it starts working & Successfully Flashed

[image]

kaangokdemir, 2019-10-23:
MacOS Catalina 10.15
balenaEtcher v1.5.59

I'm facing with that screen too.

[image]

fallenprogrammr, 2019-10-24:
> After updated to Catalina. I have reinstalled i open up. It says macos says its a mailware. So i go to System Preferences >> Security & Privacy >> Open Anyway for Etcher Balena
> And Try to reopen [...]

This is working for me.

louisgex, 2019-10-24:
@fallenprogrammr I did exactly as you said and I have an other crash message:
[image]

and it's say : 

[image]

thanks for your help

louisgex, 2019-10-25:
> After updated to Catalina. I have reinstalled i open up. It says macos says its a mailware. So i go to System Preferences >> Security & Privacy >> Open Anyway for Etcher Balena
> And Try to reopen [...]

I did exactly as you said and I have an other crash message:

[image]

and it say : 

[image]

thanks for your help

[73 more replies]

closed 2019-11-06 by zvin