balena-io/etcher #2835: Error when creating Linux boot images.

sohrob, 2019-06-26:
- **Etcher version:** 1.5.50-x64.AppImage
- **Operating system and architecture:** Both Debian 64-bit and macOS latest edition
- **Image flashed:** Trying to flash various Linux install ISOs.
- **Do you see any meaningful error information in the DevTools?** I attached my log. Not sure if that helps.

I keep getting a message saying the following after verification step:

"Something went wrong. If it is a compressed image, please check that the archive is not corrupted."
[-1561517786134.log](https://github.com/balena-io/etcher/files/3328019/-1561517786134.log)

mrpaulashbrook, 2019-06-26:
I get the same error at the same stage when trying to write a Raspbian .img image to an SD card, on a Mac running macOS 10.15 beta.

lurch, 2019-06-26:
I dunno the cause of your error, but I see in your log you have:
`Verifying 1 devices, 64% at 617.4 MB/s (total 617.4 MB/s)`
I highly doubt that your device is _really_ being read at that speed, and I think that's another example of https://github.com/balena-io/etcher/issues/1523#issuecomment-501344373

zvin, 2019-06-28:
@lurch this is an unrelated issue.
@sohrob it looks like `sudo` or the elevated command printed `0\n` on its `stderr` and we consider this as an error. Not sure how this could happen. Can you please try with etcher 1.5.51 ?

DG12, 2019-06-30:
Might this be "Long directory/filename seems to fail" #2843

lurch, 2019-07-01:
@DG12 How can that be related to #2843 when that's specifically concerned with Mac OS Catalina, and @sohrob here is running the AppImage version of Etcher, which means he must be using Linux ? :confused:

adamphetamine, 2019-07-03:
Not sure if this is related, but on Mac v1.5.33, I am also having trouble writing a bootable linux image. Worked ok with Raspbian Buster but my app requires Jessie. Downloaded Jessie and it won't work. Clues-
1. Doesn't appear to properly unmount the SD card at the end
2. Mac OS pops up saying it can't read the card, would I like to initialise it?
3. Mac OS thinks the SD card has an 'Apple Partition Map' partitioning scheme

zvin, 2019-07-03:
@adamphetamine where did you got the Jessie image from? Have you tries with another sd card /usb drive ?

adamphetamine, 2019-07-03:
more info- I was able to get this to work with a different ISO, and after updating Etcher to 1.5.51 and allowing Etcher full access in System preferences/ Security & Privacy. 
Sorry I don't know if it was the ISO that was the problem 
Worked- 2018-03-13-raspbian-stretch.zip
Did not work- 2019-04-11-rpd-x86-stretch.iso

I may have more time to diagnose tomorrow. Also note one of these was compressed, the other not...

rasyadhs, 2019-07-05:
Have you tried this solution? #2833  [https://github.com/balena-io/etcher/issues/2833#issuecomment-505145234](url)
This lazy steps work for me.
> **Workaround:**
> Moving your image inside `balenaEtcher.app` seems to do the trick.
> You can do this by selecting "Show Package Contents" on `balenaEtcher.app` and dragging your image inside of it. Then you can select your image as usual from Finder and the flashing will go ahead. **Make sure you select the image inside `balenaEtcher.app`**

adamphetamine, 2019-07-05:
sorry I forgot to add that- yes I did try that solution but it didn't work for me. I guess that makes it more likely my image was corrupt in some way. But I did have success with an older image as described above, thanks!

superbowl0, 2019-07-05:
I get the same reading...

lurch, 2019-07-05:
`2018-03-13-raspbian-stretch.zip` is an image designed to be written onto an SD card, for booting on a Raspberry Pi.
`2019-04-11-rpd-x86-stretch.iso` is an image designed to be written onto a DVD or USB stick, for booting on a PC or Mac.

So depending on what it was you were trying to achieve, it doesn't seem surprising that one or the other "didn't work" :wink:

adamphetamine, 2019-07-05:
Etcher is supposed to work with both formats if you read the docs.

lurch, 2019-07-05:
Sorry, I wasn't sure what you meant by "it won't work".

I _suspect_ (and this is just a guess, as I don't have a Mac myself) some of the problems with MacOS being unable to read the contents of a USB stick flashed with the rpd-x86 ISO might be similar to https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos (i.e. the USB stick might still boot your Mac, even if MacOS thinks the card isn't readable?)

And to clarify, I'm only a "keen fan" of Etcher trying to help out where I can, I'm not an Etcher developer. Apologies if I've given the wrong impression or caused any offence.

adamphetamine, 2019-07-05:
Hi @lurch I apologise if I was short- going away tomorrow and trying to finish off a few things makes my responses to forums a bit brief. I'll check out that link asap
Thanks

sohrob, 2019-07-10:
So I am now using the newest edition of Balena Etcher 1.5.51 and everything appears to be back to normal. Thank you for all your help and attention on this matter!

closed 2019-07-10 by sohrob

blesspronexis, 2019-10-13:
I have the same issue on the Latest MacOS Catalina 10.15. The solution is to run balenaEtcher.app/Contents/MacOS/balenaEtcher as sudo.

christophedh, 2019-10-13:
worked perfectly for me thanks

chanpaul1234572, 2019-10-13:
> I have the same issue on the Latest MacOS Catalina 10.15. The solution is to run balenaEtcher.app/Contents/MacOS/balenaEtcher as sudo.

You saved me a day thanks a lot!

lurch, 2019-10-14:
> I have the same issue on the Latest MacOS Catalina 10.15

See #2833

balgamat, 2019-10-16:
> I have the same issue on the Latest MacOS Catalina 10.15. The solution is to run balenaEtcher.app/Contents/MacOS/balenaEtcher as sudo.

Yep, that does it on Catalina. Thanks a bunch!

m0t0rh3ad, 2019-10-20:
Same issue Mac OS Catalina. I've tried different images

lurch, 2019-10-20:
For issues with Etcher on macOS Catalina, see #2833

leonardyhuang, 2019-10-21:
> I have the same issue on the Latest MacOS Catalina 10.15. The solution is to run balenaEtcher.app/Contents/MacOS/balenaEtcher as sudo.

Thanks @blesspronexis !

JGallardo, 2019-11-17:
"ThE sOlutIon iS to rUN the..." is not helpful, why not just write

```
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
```

lurch, 2019-11-17:
Apparently the issue with Catalina is already fixed in the latest versions of Etcher https://github.com/balena-io/etcher/issues/2833#issuecomment-550109402
No need for the `sudo` workaround any more.

LanceHaverkamp, 2020-03-20:
As of March 2020, this is still not fixed in Linux.  It asks for the root password, then fails.

AlexisSniffer, 2020-06-23:
Install packege repository https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 for linux

unixrh, 2020-07-01:
I guess the key problem is about etcher version. I met the same issue with v1.5.45, after updating to latest( brew cask install balenaetcher) v1.5.100 it worked well.

bee14159265358979323846, 2020-08-04:
I was on an older version 1.5.52 (er something). Recently on Mac os I had wiped, reinstalled and restored from time machine, including apps (etcher). Some apps needed permissions again, Etcher was one of them. However, giving full drive access in Privacy didn't fix the problem. Installed latest etcher, turned off full drive access, got the error. Turned full drive access on and it worked.

Kayx23, 2020-09-16:
> Apparently the issue with Catalina is already fixed in the latest versions of Etcher [#2833 (comment)](https://github.com/balena-io/etcher/issues/2833#issuecomment-550109402)
> No need for the `sudo` workaround any more.

As of Sept 16, 2020, the latest release is still 1.5.109 (from the pre-made installer) for macOS and I'm on Catalina. I tried sudo opening it and still got the "Something went wrong. If it is a compressed image, please check that the archive is not corrupted" error. 

The solution would be [this comment in issue #2833](https://github.com/balena-io/etcher/issues/2833#issuecomment-505145234)

For a detailed step-by-step: 

First you right click on Etcher to select "Show Package Content"
[image]

Then you drag and drop your iso image inside 
[image]

 Go back to Etcher and select this iso image from within the Etcher package folder 

...et voila :)

frankimhof, 2020-10-14:
> As of Sept 16, 2020, the latest release is still 1.5.109 (from the pre-made installer) for macOS and I'm on Catalina. I tried sudo opening it and still got the "Something went wrong. If it is a [...]

This worked for me! All the other solutions didn't work. Thank you!

tomasrojasc, 2020-10-23:
This worked for me too! thanks @Kayx23 !!! all the above solutions failed for me :(

pchudleigh, 2020-11-18:
In case it helps anyone with V1.5.109 on OS/X Big Sur v11.0.1 the solution that worked for me was to (after installing Etcher into Applications).

```
sudo su -
cd /Applications/balenaEtcher.app/Contents/MacOS/
./balenaEtcher
```
Then choose the downloaded ISO from my Downloads directory (/Users/$USER/Downloads) and flash that.

As documented above for Catalina.

NY3Ranger, 2020-11-26:
On High Sierra using balena 1.5.1111 and ubuntu 20.04LTS iso from their site. I tried both copying the iso to the balena folder as well as running the app sudo and they both didn't work. I still got:

Attention
Something went wrong. If it is a compressed image, please check that the archive is not corrupt. The writer process ended unexpectedly.

Any other thoughts?

tomasrojasc, 2020-11-26:
@NY3Ranger did you reset the computer. For me copying the image to the balena folder worked, I think I had to close and reopen the aplication, maybe try a system reboot just in case, I was using High Sierra.

maziyarpanahi, 2020-11-27:
This problem still exists on macOS Big Sur and 1.5.111 version of Etcher. However, either copying the image inside `balenaEtcher.app/Contents/` or running it with `sudo` as a workaround works without any issue.