balena-io/etcher #2843: Mac OS Catalina 10.15 seems to be PREVENTING writing to SD

DG12, 2019-06-30:
- **Etcher version:Version = 1.5.39,** 
- **Operating system and architecture:Mac OS Catalina 10.15 beta (19A487m)** 
From devtools; console:

Sun Jun 30 2019 08:33:44 GMT-0400 (EDT) Open image selector ({“applicationSessionUuid”:“c8c7003a-5741-4484-b98c-4f37fe3cabb6”,“flashingWorkflowUuid”:“521ab4a8-f281-4360-8af3-6964b792d80a”,“sample”:0.1})
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Sun Jun 30 2019 08:33:50 GMT-0400 (EDT) Select image ({“image”:{“size”:2197815296,“compressedSize”:426031606,“logo”:false,“name”:“2019-06-20-raspbian-buster-lite.img”,“hasMBR”:true,“partitions”:[{“offset”:4194304,“size”:268435968,“type”:12,“index”:1},{“offset”:276824064,“size”:1920991232,“type”:131,“index”:2}],“path”:"/Volumes/DATA/dgerman/Documents/computerstuff/hardware/pi/SDimages/i/2019-06-20-raspbian-buster-lite.zip",“extension”:“zip”,“blockMap”:false},“applicationSessionUuid”:“c8c7003a-5741-4484-b98c-4f37fe3cabb6”,“flashingWorkflowUuid”:“521ab4a8-f281-4360-8af3-6964b792d80a”,“sample”:0.1})
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Sun Jun 30 2019 08:33:53 GMT-0400 (EDT) Flashing 0 devices, 0% at null MB/s (total null MB/s) eta in undefineds with 0 failed devices
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Sun Jun 30 2019 08:33:53 GMT-0400 (EDT) Flash ({“image”:"/Volumes/DATA/dgerman/Documents/computerstuff/hardware/pi/SDimages/i/2019-06-20-raspbian-buster-lite.zip",“drives”:[{“size”:3904897024,“isVirtual”:false,“enumerator”:“DiskArbitration”,“logicalBlockSize”:512,“raw”:"/dev/rdisk2",“error”:null,“isReadOnly”:false,“displayName”:"/dev/disk2",“blockSize”:512,“isSCSI”:false,“isRemovable”:true,“device”:"/dev/disk2",“busVersion”:null,“isSystem”:false,“busType”:“USB”,“isCard”:true,“isUSB”:true,“devicePath”:“IODeviceTree:/PCI0@0/XHC1@14”,“mountpoints”:[{“path”:"/Volumes/boot",“label”:“boot”}],“description”:“APPLE SD Card Reader Media”,“isUAS”:null}],“driveCount”:1,“uuid”:“814bc630-f97e-44aa-8e1d-6f0df588058f”,“status”:“started”,“flashInstanceUuid”:“814bc630-f97e-44aa-8e1d-6f0df588058f”,“unmountOnSuccess”:true,“validateWriteOnSuccess”:true,“applicationSessionUuid”:“c8c7003a-5741-4484-b98c-4f37fe3cabb6”,“flashingWorkflowUuid”:“521ab4a8-f281-4360-8af3-6964b792d80a”,“sample”:0.1})
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Elevating command: /Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper.app/Contents/MacOS/balenaEtcher Helper /Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Sun Jun 30 2019 08:33:53 GMT-0400 (EDT) SafeWebview loaded ({“event”:{“fromCache”:false,“headers”:{“Accept”:“text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,/;q=0.8”,“Upgrade-Insecure-Requests”:“1”,“User-Agent”:“Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_0) AppleWebKit/537.36 (KHTML, like Gecko) balena-etcher/1.5.39 Chrome/66.0.3359.181 Electron/3.1.9 Safari/537.36”},“id”:46,“method”:“GET”,“referrer”:"",“resourceType”:“mainFrame”,“responseHeaders”:{“access-control-allow-origin”:["*"],“age”:[“0”],“cache-control”:[“max-age=600”],“cf-ray”:[“4ef02bd67c71ca6f-YUL”],“content-encoding”:[“br”],“content-type”:[“text/html; charset=utf-8”],“date”:[“Sun, 30 Jun 2019 12:33:53 GMT”],“expect-ct”:[“max-age=604800, report-uri=“https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct””],“expires”:[“Sun, 30 Jun 2019 12:43:53 GMT”],“last-modified”:[“Fri, 28 Jun 2019 14:19:32 GMT”],“server”:[“cloudflare”],“status”:[“200”],“vary”:[“Accept-Encoding”],“via”:[“1.1 varnish”],“x-cache”:[“MISS”],“x-cache-hits”:[“0”],“x-fastly-request-id”:[“aff951f4eface37170ff8cd342a254f6214faab7”],“x-github-request-id”:[“1B46:09FF:2B4578:383602:5D18AC30”],“x-proxy-cache”:[“MISS”],“x-served-by”:[“cache-yul8922-YUL”],“x-timer”:[“S1561898034.714297,VS0,VE22”]},“statusCode”:200,“statusLine”:“HTTP/1.1 200”,“timestamp”:1561898033742.375,“url”:“https://assets.balena.io/etcher-featured/index.html?etcher-version=1.5.39&api-version=2&optOutAnalytics=false"},“applicationSessionUuid”:“c8c7003a-5741-4484-b98c-4f37fe3cabb6”,“flashingWorkflowUuid”:“521ab4a8-f281-4360-8af3-6964b792d80a”,"sample”:0.1})
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Sun Jun 30 2019 08:33:53 GMT-0400 (EDT) Flashing 0 devices, 0% at null MB/s (total null MB/s) eta in undefineds with 0 failed devices
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Successfully connected to IPC server: etcher-server-23631, socket root /var/folders/rr/jvwb6c9j2h18lzc28kzws_940000gn/T/
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Image: /Volumes/DATA/dgerman/Documents/computerstuff/hardware/pi/SDimages/i/2019-06-20-raspbian-buster-lite.zip
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Devices: /dev/disk2
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Umount on success: true
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Validate on success: true
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Error: EPERM: operation not permitted, open ‘/Volumes/DATA/dgerman/Documents/computerstuff/hardware/pi/SDimages/i/2019-06-20-raspbian-buster-lite.zip’
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Sun Jun 30 2019 08:33:57 GMT-0400 (EDT) Open modal ({“name”:“warning”,“applicationSessionUuid”:“c8c7003a-5741-4484-b98c-4f37fe3cabb6”,“flashingWorkflowUuid”:“521ab4a8-f281-4360-8af3-6964b792d80a”,“sample”:0.1})
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

{“stack”:“Error: EPERM: operation not permitted, open ‘/Volumes/DATA/dgerman/Documents/computerstuff/hardware/pi/SDimages/i/2019-06-20-raspbian-buster-lite.zip’”,“message”:“EPERM: operation not permitted, open ‘/Volumes/DATA/dgerman/Documents/computerstuff/hardware/pi/SDimages/i/2019-06-20-raspbian-buster-lite.zip’”,“name”:“Error”,“code”:“EPERM”,“syscall”:“open”,“errno”:-1,“image”:“2019-06-20-raspbian-buster-lite.zip”}
console.(anonymous function) @ /Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Flash results Object
/Volumes/DATA/dgerman/.bin/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

Terminating IPC server

novemTeam, 2019-07-03:
I had the same issue on macOS Catalina, I moved the file to my home folder and renamed the file to a shorter name. This did indeed do the trick for me.

semicolonsnet, 2019-07-06:
I had to run as "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher" to get correct permissions to write to the SD card. Running the app normally didn't seem to work.

bk-one, 2019-07-08:
same here, running as sudo did the trick, it gives you an additional dialog and asks if it's okay for Etcher to write to a removable disk. Didn't get this dialog when running as my user.

idoodler, 2019-07-21:
I don't get any dialog, running with `sudo` didn't help. I also enabled the new Filesystem permissions in the Settings app. I am now using Parallels Dektop as a Workaround.

jsajovic, 2019-07-25:
@idoodler 

As long as you are logged into your Mac as an Admin user you will have access to sudo. As @semicolonsnet  mentioned Open a terminal window and navigate all the way into the contents of the application and then open the app. Here is a copy of my output for reference.  `MacBook-Pro:MacOS jsajovic$ sudo ./balenaEtcher
Password:
2019-07-25 12:19:16.942 balenaEtcher[14093:2684760] NSSoftLinking - The function 'SLSIsSuppressedByScreenTime' can't be found in the (null) framework.
2019-07-25 12:19:17.014 balenaEtcher[14093:2684760] *** WARNING: Textured window <AtomNSWindow: 0x7fcb943f06d0> is getting an implicitly transparent titlebar. This will break when linking against newer SDKs. Use NSWindow's -titlebarAppearsTransparent=YES instead.
ready-to-show: 2156.483ms
Checking for update
Update for version 1.5.52 is not available (latest version: 1.5.52, downgrade is disallowed).
Error: Error: Please check update first
    at MacUpdater.downloadUpdate (/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:576:21)
    at checkForUpdates (/Applications/balenaEtcher.app/Contents/Resources/app.asar/generated/etcher.js:435:27)
    at process._tickCallback (internal/process/next_tick.js:68:7)
{"stack":"Error: Please check update first\n    at MacUpdater.downloadUpdate (/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:576:21)\n    at checkForUpdates (/Applications/balenaEtcher.app/Contents/Resources/app.asar/generated/etcher.js:435:27)\n    at process._tickCallback (internal/process/next_tick.js:68:7)","message":"Please check update first"}
{"stack":"Error: Please check update first\n    at MacUpdater.downloadUpdate (/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:576:21)\n    at checkForUpdates (/Applications/balenaEtcher.app/Contents/Resources/app.asar/generated/etcher.js:435:27)\n    at process._tickCallback (internal/process/next_tick.js:68:7)","message":"Please check update first"}
Checking for update
Update for version 1.5.52 is not available (latest version: 1.5.52, downgrade is disallowed).
Error: Error: Please check update first
    at MacUpdater.downloadUpdate (/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:576:21)
    at checkForUpdates (/Applications/balenaEtcher.app/Contents/Resources/app.asar/generated/etcher.js:435:27)
    at process._tickCallback (internal/process/next_tick.js:68:7)
{"stack":"Error: Please check update first\n    at MacUpdater.downloadUpdate (/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:576:21)\n    at checkForUpdates (/Applications/balenaEtcher.app/Contents/Resources/app.asar/generated/etcher.js:435:27)\n    at process._tickCallback (internal/process/next_tick.js:68:7)","message":"Please check update first"}
{"stack":"Error: Please check update first\n    at MacUpdater.downloadUpdate (/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:576:21)\n    at checkForUpdates (/Applications/balenaEtcher.app/Contents/Resources/app.asar/generated/etcher.js:435:27)\n    at process._tickCallback (internal/process/next_tick.js:68:7)","message":"Please check update first"}
`

idoodler, 2019-07-25:
@jsajovic Yes, I know that. Executing `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher` didn't help, the Application starts as expected, but I can't flash the image on the SD Card. There is no error message in the applications log.

DG12, 2019-07-25:
@idoodler What directory is the image you want to flash in?
Did you get the "something went wrong" message?

idoodler, 2019-07-26:
@DG12 The image is in the `~/Downloads` directory and yes, I get the "something went wrong" message but without any further information and without any error in the terminal window.

DG12, 2019-07-26:
@idoodler Try moving the image file to /tmp or /private/tmp .
Let us know what happens.

idoodler, 2019-07-27:
@DG12 No, same error message in the BalenaEtcher application. I executed the binary with sudo...
[image]

DG12, 2019-07-27:
This message: "The writer process ended unexpectedly" in not the " "something went wrong"  message you said you received earlier.

idoodler, 2019-07-27:
@DG12 Oh, I thought you ment this message because you its less to write.

thundron, 2019-07-29:
@idoodler could you please paste here the output of the devtools console? `cmd + shift + i` to open it

redsox4886, 2019-08-02:
Moving the linux iso to my user folder and then running the app under sudo (sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher) did the trick.  I did still get the new Catalina prompt asking to confirm access to the USB drive but no further issues. 

Thanks for the tips!

f1rat, 2019-08-30:
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher worked great for me. I also renamed my image file to a.img and left it in the Downloads folder.

OSX 10.15 19A546d with Etcher 1.5.56

DG12, 2019-09-23:
To repeat my request which will assist in isolating the specific problem:
PLEASE revise the error message ""Something went wrong" when "operation not permitted, open " occurs to be more specific rather than mentioning corruption.
Thank you.

thundron, 2019-09-25:
@DG12 We'll do, there are some new error messages we need to flesh out when the generic "EPERM" error appears, and in this case it's due to the changes to OSX Catalina. See https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047 for a workaround and let's use that as the issue for Catalina changes, closing this one

closed 2019-09-25 by thundron

felixlindemann, 2019-10-12:
Same here.... Running as sudo worked for me.
thanks to @semicolonsnet

bee256, 2019-10-12:
Same with me. Working with balenaEtcher 1.5.58 on macOS Catalina 10.15. It worked only when run as sudo.

mattbeau, 2019-10-13:
running as sudo is working for me also thanks @semicolonsnet

wAuner, 2019-10-13:
thanks, `sudo` worked for as well. Before I got the error message "something went wrong" and something about a compressed image.
`sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`

keiohtani, 2019-10-13:
I tried with balenaEtcher version 1.5.30 with the command `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`, and it did not work. 

After updating the app to version 1.5.58, the app is working fine now!

marcosanchez, 2019-10-16:
I had the same problem, and worked for me the sudo solution. Maybe something interesting is that with this method I had to allow access to the USB drive for terminal.

rajkundu, 2019-10-17:
Thank you @semicolonsnet! Worked beautifully for me.

DIYglenn, 2019-10-22:
> I had to run as "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher" to get correct permissions to write to the SD card. Running the app normally didn't seem to work.

Great, thanks!

diver-down, 2019-10-29:
Same, running `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher` within terminal launches the app with the proper permissions.

That said. This feels a bit unsafe.

lurch, 2019-10-29:
@diver-down It's being fixed in #2939 
See #2833 for more details.

BillSquare, 2019-10-31:
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
That worked for me :-)
Txxxxx
does somebody can explain why this would be unsafe ?

3DDruckEinsteiger, 2019-10-31:
"sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher" worked!

KMDbogo, 2019-11-05:
> "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher" worked!

Thx !!

michaelschapira, 2019-11-14:
I am on macOS 10.15.1 and I had to `sudo` the application to flash a device as well. 
I was getting a "corrupted image" error message.

lurch, 2019-11-15:
According to https://github.com/balena-io/etcher/issues/2833#issuecomment-550109402 this should be fixed in the latest versions of Etcher. (making `sudo` no longer necessary)

DG12, 2019-11-15:
Catalina 10.15.2 this is not a problem

RauchenwaldC, 2019-12-16:
> "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher" worked!

Balena Etcher 1.5.69 on macOS 10.15.1 works with sudo. Thanks.

WorldWideWebDev, 2021-01-11:
Same problem on Balena 1.5.113  I inadvertently clicked "Disallow"  a Mac Request to allow Balena to access my SD card and then i received an error from Balena "Writing to card failed,  EPREM error Permission Denied " or similar. I then tried "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher " from Terminal and started Balena once again. I then received a request from my Mac " Terminal wants to access / write to SD card" i clicked on "Allow" at this stage and all is well.  Seems it was an ID-10-T error on my part.

TimidRobot, 2021-05-03:
I encoutered this error on macOS Big Sur 11.3

I resolved it by ensuring System Preferences > Security & Privacy > Privacy > Files and Folders > Removal Volumes was enabled for balenaEtcher